NMGRL/pychron #1115: flux calculation quit - pipeline canceled by Fit Flux

wcmcintosh, 2017-05-23:
'''
Pipeline Data Processing                : 2017-05-23 09:52:16,604 DEBUG     (MainThread) run needed for None
Pipeline Data Processing                : 2017-05-23 09:52:16,604 DEBUG     (MainThread) run pipeline started
pychronmeta                             : 2017-05-23 09:52:16,604 DEBUG     (MainThread) create new session 4975384912
PipelineEngine                          : 2017-05-23 09:52:16,606 DEBUG     (MainThread) pipeline run started
2017-05-23 09:52:18.685 python[13615:2033911] modalSession has been exited prematurely - check for a reentrant call to endModalSession:
pychronmeta                             : 2017-05-23 09:52:18,723 DEBUG     (MainThread) SELECT `AnalysisTbl`.id, `AnalysisTbl`.timestamp, `AnalysisTbl`.uuid, `AnalysisTbl`.analysis_type, `AnalysisTbl`.aliquot, `AnalysisTbl`.increment, `AnalysisTbl`.`irradiation_positionID`, `AnalysisTbl`.`measurementName`, `AnalysisTbl`.`extractionName`, `AnalysisTbl`.`postEqName`, `AnalysisTbl`.`postMeasName`, `AnalysisTbl`.mass_spectrometer, `AnalysisTbl`.extract_device, `AnalysisTbl`.extract_value, `AnalysisTbl`.extract_units, `AnalysisTbl`.cleanup, `AnalysisTbl`.duration, `AnalysisTbl`.weight, `AnalysisTbl`.comment 
FROM `AnalysisTbl` JOIN `IrradiationPositionTbl` ON `IrradiationPositionTbl`.id = `AnalysisTbl`.`irradiation_positionID` JOIN `LevelTbl` ON `LevelTbl`.id = `IrradiationPositionTbl`.`levelID` JOIN `IrradiationTbl` ON `IrradiationTbl`.id = `LevelTbl`.`irradiationID` JOIN `SampleTbl` ON `SampleTbl`.id = `IrradiationPositionTbl`.`sampleID` JOIN `AnalysisChangeTbl` ON `AnalysisTbl`.id = `AnalysisChangeTbl`.`analysisID` 
WHERE `IrradiationTbl`.name = NM-289 AND `LevelTbl`.name = E AND `SampleTbl`.name = FC-2 AND `AnalysisChangeTbl`.tag != invalid
DVC                                     : 2017-05-23 09:52:18,889 DEBUG     (MainThread) sync repository Load1795. exists=True
Inspection                              : 2017-05-23 09:52:18,890 DEBUG     (MainThread) pull called by sync_repo. parent call=<pychron.dvc.dvc.DVC object at 0x1288e57d0> repo.pull(use_progress=use_progress)
Load1795<GitRepo>                       : 2017-05-23 09:52:18,890 DEBUG     (MainThread) pulling master from origin
Load1795<GitRepo>                       : 2017-05-23 09:52:18,891 DEBUG     (MainThread) pulling from url: https://github.com/NMGRLData/Load1795.git
DVC                                     : 2017-05-23 09:52:20,225 DEBUG     (MainThread) Make analysis time, total: 1.33583283424, n: 48, average: 0.0278298507134
PipelineEngine                          : 2017-05-23 09:52:20,228 DEBUG     (MainThread) 00: Find Flux Monitors<FindFluxMonitorsNode> Runtime: 1.5419
PipelineEngine                          : 2017-05-23 09:52:20,234 DEBUG     (MainThread) 01: Flux Monitors<FluxMonitorsNode> Runtime: 0.0059
2017-05-23 09:52:24.409 python[13615:2033911] modalSession has been exited prematurely - check for a reentrant call to endModalSession:
PipelineEngine                          : 2017-05-23 09:52:24,419 DEBUG     (MainThread) 02: Fit Flux<FitFluxNode> Runtime: 0.0088
PipelineEngine                          : 2017-05-23 09:52:24,419 DEBUG     (MainThread) pipeline canceled by Fit Flux<FitFluxNode>
Pipeline Data Processing                : 2017-05-23 09:52:24,419 DEBUG     (MainThread) true run pipeline run_pipeline
Pipeline Data Processing                : 2017-05-23 09:52:25,029 DEBUG     (MainThread) run pipeline finished

;'''

jirhiker, 2017-05-23:
Use ``` not ''' if you want to include log info

jirhiker, 2017-05-23:
need more information

wcmcintosh, 2017-05-24:
All OK

closed 2017-05-24 by jirhiker