NMGRL/pychron #1121: K/Ca log plot minimum

wcmcintosh, 2017-05-23:
in Spectra, if K/Ca is on a log scale and the errors extend <0, the minimum shown on the plot is 1, even if the scale extends to <1.

Shouldn't the error bar extend to the minimum of the plot in this case?\

Not a burning issue at this point - it only affects bad data

jirhiker, 2017-05-24:
not when the user specifies the limits of the graph

wcmcintosh, 2017-05-24:
To Clarify - if the error on a K/Ca value takes it into negative space, then the bar plotted on any log scale should extend to the bottom of the graph, regardless of what the user specifies.  Currently, the bar for any negative going value extends from the maximum value down to 1.0, regardless of the minimum value set for the graph.  Am I not correct in my thinking here?

jirhiker, 2017-05-24:
What does it do for K/Ca values less than 1.0

closed 2019-03-17 by jirhiker