NMGRL/pychron #1123: Error saving plateau age as IA - will try after restart

wcmcintosh, 2017-05-24:
active branch=None

active analyses=65496-05A,65496-05B,65496-05C,65496-05D,65496-05E

```
Traceback (most recent call last):
  File "/Users/mcintosh/miniconda2/envs/pychron/lib/python2.7/site-packages/pyface/ui/qt4/action/action_item.py", line 343, in _qt4_on_triggered
    self.controller.perform(action, action_event)
  File "/Users/mcintosh/miniconda2/envs/pychron/lib/python2.7/site-packages/pyface/tasks/action/task_action_controller.py", line 31, in perform
    return action.perform(event)
  File "/Users/mcintosh/miniconda2/envs/pychron/lib/python2.7/site-packages/pyface/tasks/action/listening_action.py", line 58, in perform
    method()
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 335, in set_interpreted_age
    self._add_interpreted_ages(ias)
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 475, in _add_interpreted_ages
    dvc.add_interpreted_age(ia)
  File "/Users/mcintosh/Programming/git/pychron/pychron/dvc/dvc.py", line 768, in add_interpreted_age
    ia.all_analyses])
  File "/Users/mcintosh/Programming/git/pychron/pychron/processing/analyses/analysis_group.py", line 427, in get_is_plateau_step
    idx = self.analyses.index(an)
ValueError: <pychron.dvc.dvc_analysis.DVCAnalysis object at 0x12e986410> is not in list

```

wcmcintosh, 2017-05-24:
saving IA plateau age repeatable problem for 65496-05
others analyses OK

wcmcintosh, 2017-05-24:
this only is a problem for sample with omitted steps and a plateau, e.g. 65496-05 and -07.  I would cancel the omitted steps (a relic of previous plateau problems) except for previously noted tagging problem

jirhiker, 2017-05-24:
I concur. The issue had to do with setting IA on a spectrum with  a plateau but in which a step identified as part of the plateau was omitted by the user

wcmcintosh, 2017-05-24:
OK

closed 2017-05-24 by jirhiker