NMGRL/pychron #1125: Erro on tryuing to tag IA - not yet implemented?

wcmcintosh, 2017-05-24:
active branch=None

active analyses=

```
Traceback (most recent call last):
  File "/Users/mcintosh/miniconda2/envs/pychron/lib/python2.7/site-packages/pyface/ui/qt4/action/action_item.py", line 343, in _qt4_on_triggered
    self.controller.perform(action, action_event)
  File "/Users/mcintosh/miniconda2/envs/pychron/lib/python2.7/site-packages/pyface/tasks/action/task_action_controller.py", line 31, in perform
    return action.perform(event)
  File "/Users/mcintosh/miniconda2/envs/pychron/lib/python2.7/site-packages/pyface/tasks/action/listening_action.py", line 58, in perform
    method()
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 254, in set_tag
    if dvc.update_tag(it):
  File "/Users/mcintosh/Programming/git/pychron/pychron/dvc/dvc.py", line 445, in update_tag
    tag = Tag.from_analysis(an)
  File "/Users/mcintosh/Programming/git/pychron/pychron/dvc/dvc.py", line 72, in from_analysis
    tag.path = analysis_path(an.record_id, an.repository_identifier, modifier='tags')
  File "/Users/mcintosh/Programming/git/pychron/pychron/dvc/__init__.py", line 92, in analysis_path
    raise AnalysisNotAnvailableError(root, runid)
AnalysisNotAnvailableError: Analysis Not Available. /Users/mcintosh/Pychron_dev/data/.dvc/repositories/ia - 65495 -15x

```

wcmcintosh, 2017-05-24:
tagging in fact worked in spite of error

wcmcintosh, 2017-05-24:
actually, tagging only tagged first selection, then tag was not saved and plot not updated

jirhiker, 2017-05-24:
step 1 is to stop using '-' and spaces in IA names

wcmcintosh, 2017-05-24:
i was using -, I will switch to _
I don't think I used any spaces

wcmcintosh, 2017-05-24:
all better now

closed 2017-05-24 by jirhiker