NMGRL/pychron #1218: Default blank fits

wcmcintosh, 2017-11-02:
Default preceding  blank fits at the time of analysis are using air and cocktail blanks for unknowns.  Is there a setting to avoid this?

jirhiker, 2017-11-02:
no setting at this point. its up to the user to set up the experiment properly

wcmcintosh, 2018-06-22:
I know that this issue has been around for a long time, but I still wonder why not take the previous appropriate blank, rather than any old previous blank.

jirhiker, 2018-06-22:
See previous statement

jirhiker, 2018-06-22:
To NMGRL setting up an experiment with correct proceeding blanks would result in excessive blanks. For other labs this is SOP. 

To explain why this feature is not being adding
1. The process of setting the preceding blank is already complicated
2. Going to the model you suggests requires a change in the data structure used to track the preceding blank.
3. The change in data structure will have cause a cascade of errors that will need to be resolved on the data collection machines
4. Ultimately not necessary as blanks are almost always refit and the blank fit pipeline can respect the analysis type association

closed 2019-03-17 by jirhiker