NMGRL/pychron #1256: improve ability to distinguish data on multi-sample plot

wcmcintosh, 2018-01-17:
could you add a legend or other means  to help distinguish results from multiple samples in ideograms or spectra?

currently samples and L#s can be listed in the title, but not readily tied to the different color data points

this could be addressed with a legend

alternatively, if one could add the sample # or L# to the mean age string, this might work well too

wcmcintosh, 2018-05-26:
Not sure on status of this one.  I had thought that an ideogram legend had been added, but I am not finding it in options.

jirhiker, 2018-05-26:
"sample # or L# to the mean age string" this was implemented

wcmcintosh, 2018-06-22:
Labels on mean ages work with one small exception.  If Identifier(Sample) is selected, then the word "Identifier" appears instead of the L#.  If just "Identifier" or "Identifier Sample" is selected, the L# appears in the mean age label.

closed 2019-03-17 by jirhiker