NMGRL/pychron #1309: SD vs SEM

nmgrluser, 2018-03-20:
In fitting IC data to some NM-281 data, using linear fit, I note that SD errors are ±0.0044 and that SEM errors are ±0.000007, roughly three orders of magnitude smaller.  The number of references is 13, so shouldn't SEM=SD/squareroot (13)  or SEM=SD/3.6? At least approximately? Am I missing something here?

nmgrluser, 2018-03-20:
I note that Cl errors are roughly half of SD errors.

nmgrluser, 2018-03-20:
If average fit is selected, SD and SEM seem to scale appropriately for n

nmgrluser, 2018-03-20:
Is the  SEM/SD discrepency an artifact of how errors are calculated for line fits?

jirhiker, 2018-03-21:
SEM=SD/root(N) does not apply to linear regression. that is eq is relevant when calculating the mean. 

error calc for Ordinary Least Squares regression follows Draper and Smith chapter 2.4 page 56

jirhiker, 2018-03-21:
Also as ive mentioned numerous times the OLS statistics are not applicable to time series and different approach is needed for predicting the error at  a given point in time

closed 2018-03-23 by jirhiker