NMGRL/pychron #1326: browser window tagging

nmgrluser, 2018-04-03:
Is there any chance that you would like to reconsider tagging from the browser window analysis list?  

It seems like tagging in some cases would pair exceptionally well with the ability to select by attribute

I get your argument about not having info in the browser stage to guide tagging, but in reality, we often circle back to the browser repeatedly in the course of data workflow

nmgrluser, 2018-04-04:
Testing #1326, tagging as skip triggered this exception

root                                    : 2018-04-04 15:59:15,338 CRITICAL  (MainThread) Traceback (most recent call last):
root                                    : 2018-04-04 15:59:15,338 CRITICAL  (MainThread) File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.4/site-packages/pyface/ui/qt4/action/action_item.py", line 163, in _qt4_on_triggered
    self.controller.perform(action, action_event)
root                                    : 2018-04-04 15:59:15,339 CRITICAL  (MainThread) File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.4/site-packages/traitsui/qt4/editor.py", line 269, in perform
    self.ui.do_undoable( self._perform, action )
root                                    : 2018-04-04 15:59:15,339 CRITICAL  (MainThread) File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.4/site-packages/traitsui/ui.py", line 716, in do_undoable
    action( *args, **kw )
root                                    : 2018-04-04 15:59:15,339 CRITICAL  (MainThread) File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.4/site-packages/traitsui/qt4/editor.py", line 291, in _perform
    method( info, selection )
root                                    : 2018-04-04 15:59:15,339 CRITICAL  (MainThread) File "/Users/argonlab2/.pychron.1/updates/pychron/envisage/browser/sample_view.py", line 370, in tag_skip
    self._set_tags(info.object, obj, 'skip')
root                                    : 2018-04-04 15:59:15,339 CRITICAL  (MainThread) TypeError: _set_tags() takes 3 positional arguments but 4 were given
root                                    : 2018-04-04 15:59:15,339 CRITICAL  (MainThread) ============ End Exception ==========

nmgrluser, 2018-04-04:
works fine to tag as omit

wcmcintosh, 2018-04-06:
All OK now, but, as Jake warned, database interaction on tagging is very slow from remote locations 
See new ticket on this

closed 2018-04-12 by jirhiker