NMGRL/pychron #1330: database interaction on tagging speed - not show stopper

wcmcintosh, 2018-04-06:
I was warned on this, but am wondering if there is room for optimization

Currently each batch of tagging updates triggers a sequence of updates for each tag.  Updating each analysis triggers the following output, taking 30-60 seconds per analysis.

Can these each batch of database updates be handled as a batch, rather than individual analyses?

GitRepoManager                          : 2018-04-05 15:38:45,664 DEBUG     (MainThread) pull complete
ISO                                     : 2018-04-05 15:38:46,263 WARNING   (MainThread)  - Failed loading peakcenter. error='int' object is not iterable
DVC                                     : 2018-04-05 15:38:46,295 DEBUG     (MainThread) Make analysis time, total: 2.646667957305908, n: 1, average: 2.646667957305908
DVC                                     : 2018-04-05 15:38:46,296 DEBUG     (MainThread) setting 64110-09A tag= skip
GitRepoManager                          : 2018-04-05 15:38:49,547 DEBUG     (MainThread) add paths /Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/10-09A.tags.json
git.cmd                                 : 2018-04-05 15:38:49,548 DEBUG     (MainThread) Popen(['git', 'diff', '--abbrev=40', '--full-index', '-M', '--raw', '--no-color'], cwd=/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba, universal_newlines=False, shell=None)
git.cmd                                 : 2018-04-05 15:38:49,618 DEBUG     (MainThread) Popen(['git', 'status', '--porcelain', '--untracked-files'], cwd=/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba, universal_newlines=False, shell=None)
aa []
bb []
tt ['/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/ia/09.ia.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/07-01A.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/07-02A.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/07-03A.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/07-04A.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/07-05A.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/07-06A.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/07-07A.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/07-08A.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/07-09A.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/07-10A.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/07-11A.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/07-12A.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/07-13A.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/09-01A.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/09-01B.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/09-01C.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/09-02A.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/09-02C.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/09-03A.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/09-04B.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/09-04C.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/09-05A.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/09-05B.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/09-05C.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/09-05D.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/09-05E.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/09-05F.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/09-05G.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/09-06A.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/09-06B.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/09-06E.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/10-01A.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/10-02A.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/10-03A.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/10-04A.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/10-05A.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/10-06A.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/10-07A.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/10-08A.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/10-09A.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/10-10A.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/10-11A.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/10-11C.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/10-12A.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/10-12C.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/10-13A.tags.json', '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba/641/tags/10-13C.tags.json']
ps []
DVC                                     : 2018-04-05 15:38:50,201 DEBUG     (MainThread) sync repository Toba. exists=True
Inspection                              : 2018-04-05 15:38:50,206 DEBUG     (MainThread) pull called by sync_repo. parent call=<pychron.dvc.dvc.DVC object at 0x123375d58> repo.pull(use_progress=use_progress)
GitRepoManager                          : 2018-04-05 15:38:50,206 DEBUG     (MainThread) pulling master from origin
GitRepoManager                          : 2018-04-05 15:38:50,208 DEBUG     (MainThread) pulling from url: https://github.com/NMGRLData/Toba.git
git.cmd                                 : 2018-04-05 15:38:50,209 DEBUG     (MainThread) Popen(['git', 'fetch', 'origin'], cwd=/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba, universal_newlines=False, shell=None)
GitRepoManager                          : 2018-04-05 15:38:52,092 DEBUG     (MainThread) fetch complete
git.cmd                                 : 2018-04-05 15:38:52,093 DEBUG     (MainThread) Popen(['git', 'merge', 'FETCH_HEAD'], cwd=/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba, universal_newlines=False, shell=None)
GitRepoManager                          : 2018-04-05 15:38:52,153 DEBUG     (MainThread) pull complete

wcmcintosh, 2018-04-06:
speed of course depends on internet.  more like 8 sec per sample this morning

wcmcintosh, 2018-04-06:
tagging from the analysis pane does not seem to trigger a database push (update?)

jirhiker, 2018-04-06:
the database is never pushed. and analysis repositories are never pushed automatically.

closed 2018-04-06 by jirhiker