NMGRL/pychron #1358: sigma and ± symbols not working in illustrator

mjz1983, 2018-04-12:
For arial, the sigma symbol is correctly shown in illustrator the the "±" is not.

For helvetica, neither the sigma or "±" symbol is shown correctly.

jirhiker, 2018-04-13:
Ive spent many hours on this on with essentially no success. 

I also cannot reproduce those symptoms. 
for me helvetica is the only font that works properly.
and i never had arial property display plusminus

if you open the file in Preview or Acrobat are the symbols displayed correctly? for me they are

The only solution Ive discovered is far from ideal. 
1. open the file in Acrobat. 
2. Export to PostScript. this will create a .ps file
3. Open the PostScript file Illustrator

jirhiker, 2018-04-13:
More investigation suggests no easy solution other than opening file in Acrobat and saving as one of the following

1. Encapsulated PostScript 
2. PostScript
3. PDF/A

mjz1983, 2019-03-05:
Any progress on this front? Sigma and ± are still not displayed correclty in adobe illustrator CC2018. Symbols appear correct in pychron display, but not in illusrator

jirhiker, 2019-03-05:
No and its very computer dependent. Im hoping if I can resolve issue #1719 this problem will go away, tho probably not

jirhiker, 2019-03-15:
one work around is to open in Acrobat and Save As Other > PDF/A  Then the text is ok in Illustrator