NMGRL/pychron #1371: blank fitting exception

wcmcintosh, 2018-04-16:
show stopping exception while fitting blanks for NM-274 Toba data, occurred after references identified and accepted

pychronmeta                             : 2018-04-16 11:44:07,168 DEBUG     (MainThread) SELECT "AnalysisTbl".id, "AnalysisTbl".timestamp, "AnalysisTbl".uuid, "AnalysisTbl".analysis_type, "AnalysisTbl".aliquot, "AnalysisTbl".increment, "AnalysisTbl"."irradiation_positionID", "AnalysisTbl"."measurementName", "AnalysisTbl"."extractionName", "AnalysisTbl"."postEqName", "AnalysisTbl"."postMeasName", "AnalysisTbl".mass_spectrometer, "AnalysisTbl".extract_device, "AnalysisTbl".extract_value, "AnalysisTbl".extract_units, "AnalysisTbl".cleanup, "AnalysisTbl".duration, "AnalysisTbl".weight, "AnalysisTbl".comment 
FROM "AnalysisTbl" JOIN "AnalysisChangeTbl" ON "AnalysisTbl".id = "AnalysisChangeTbl"."analysisID" 
WHERE lower("AnalysisTbl".mass_spectrometer) = 'jan' AND "AnalysisTbl".analysis_type IN ('blank_unknown') AND lower("AnalysisTbl".extract_device) IN ('fusions co2') AND "AnalysisTbl".timestamp >= '2015-04-08 05:41:41' AND "AnalysisTbl".timestamp <= '2015-04-10 02:38:21' AND "AnalysisChangeTbl".tag != 'invalid' AND "AnalysisTbl".id NOT IN (9869, 9870, 9871, 9872, 9895, 9896, 9897, 9898, 9947, 9948, 9949, 9950, 9951, 9952, 9953, 9954, 9955, 9956, 9957, 9209, 9210, 9211, 9212, 9213, 9214, 10047, 10048, 10049, 10050, 10051, 10052, 10053, 10054, 10055, 10056, 10057, 10058, 10059, 10060, 10061, 10062, 10063, 10064, 10065, 10066, 10067, 10068, 10069, 10070, 10125, 10126, 10127, 10128, 10129, 10130, 10131, 10132, 10133, 10134, 10135, 10136, 10137, 10138, 10139) ORDER BY "AnalysisTbl".timestamp ASC
2018-04-16 11:44:12.881 python[15667:23338148] modalSession has been exited prematurely - check for a reentrant call to endModalSession:
DVC                                     : 2018-04-16 11:44:13,192 DEBUG     (MainThread) sync repository Irradiation-NM-274. exists=False
DVC                                     : 2018-04-16 11:44:13,192 DEBUG     (MainThread) getting repository from remote
DVC                                     : 2018-04-16 11:44:13,192 DEBUG     (MainThread) load repositories from NMGRLData
DVC                                     : 2018-04-16 11:44:13,199 DEBUG     (MainThread) sync repository Toba. exists=True
Inspection                              : 2018-04-16 11:44:13,202 DEBUG     (MainThread) pull called by sync_repo. parent call=<pychron.dvc.dvc.DVC object at 0x11b38aca8> repo.pull(use_progress=use_progress)
GitRepoManager                          : 2018-04-16 11:44:13,202 DEBUG     (MainThread) pulling master from origin
GitRepoManager                          : 2018-04-16 11:44:13,203 DEBUG     (MainThread) pulling from url: https://github.com/NMGRLData/Toba.git
git.cmd                                 : 2018-04-16 11:44:13,204 DEBUG     (MainThread) Popen(['git', 'fetch', 'origin'], cwd=/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba, universal_newlines=False, shell=None)
GitRepoManager                          : 2018-04-16 11:44:13,802 DEBUG     (MainThread) fetch complete
git.cmd                                 : 2018-04-16 11:44:13,803 DEBUG     (MainThread) Popen(['git', 'merge', 'FETCH_HEAD'], cwd=/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Toba, universal_newlines=False, shell=None)
GitRepoManager                          : 2018-04-16 11:44:13,841 DEBUG     (MainThread) pull complete
DVC                                     : 2018-04-16 11:44:14,117 DEBUG     (MainThread) sync repository Cascades. exists=True
DVC                                     : 2018-04-16 11:44:14,117 DEBUG     (MainThread) make new repomanager for /Users/mcintosh/Pychron_dev/data/.dvc/repositories/Cascades
Inspection                              : 2018-04-16 11:44:14,124 DEBUG     (MainThread) pull called by sync_repo. parent call=<pychron.dvc.dvc.DVC object at 0x11b38aca8> repo.pull(use_progress=use_progress)
GitRepoManager                          : 2018-04-16 11:44:14,125 DEBUG     (MainThread) pulling master from origin
GitRepoManager                          : 2018-04-16 11:44:14,126 DEBUG     (MainThread) pulling from url: https://github.com/NMGRLData/Cascades.git
git.cmd                                 : 2018-04-16 11:44:14,126 DEBUG     (MainThread) Popen(['git', 'fetch', 'origin'], cwd=/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Cascades, universal_newlines=False, shell=None)
GitRepoManager                          : 2018-04-16 11:44:14,624 DEBUG     (MainThread) fetch complete
git.cmd                                 : 2018-04-16 11:44:14,624 DEBUG     (MainThread) Popen(['git', 'merge', 'FETCH_HEAD'], cwd=/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Cascades, universal_newlines=False, shell=None)
GitRepoManager                          : 2018-04-16 11:44:14,662 WARNING   (MainThread) Git command failed. Cmd('git') failed due to: exit code(128)
  cmdline: git merge FETCH_HEAD
  stderr: 'error: Merging is not possible because you have unmerged files.
hint: Fix them up in the work tree, and then use 'git add/rm <file>'
hint: as appropriate to mark resolution and make a commit.
fatal: Exiting because of an unresolved conflict.', merge
GitRepoManager                          : 2018-04-16 11:44:14,663 DEBUG     (MainThread) pull complete
root                                    : 2018-04-16 11:44:14,665 CRITICAL  (MainThread) ============ Exception ==============
root                                    : 2018-04-16 11:44:14,665 CRITICAL  (MainThread) Traceback (most recent call last):
root                                    : 2018-04-16 11:44:14,665 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traits/trait_notifiers.py", line 519, in _dispatch_change_event
    self.dispatch( handler, *args )
root                                    : 2018-04-16 11:44:14,665 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traits/trait_notifiers.py", line 482, in dispatch
    handler( *args )
root                                    : 2018-04-16 11:44:14,665 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 560, in _handle_run_needed
    self.run()
root                                    : 2018-04-16 11:44:14,665 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 308, in run
    self._run_pipeline()
root                                    : 2018-04-16 11:44:14,665 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 489, in _run_pipeline
    self._run('run pipeline', 'run_pipeline')
root                                    : 2018-04-16 11:44:14,666 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 468, in _run
    if not getattr(self.engine, func)():
root                                    : 2018-04-16 11:44:14,666 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/engine.py", line 722, in run_pipeline
    node.run(state)
root                                    : 2018-04-16 11:44:14,666 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/nodes/find.py", line 243, in run
    if self._run_group(state, gid, list(ans)):
root                                    : 2018-04-16 11:44:14,666 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/nodes/find.py", line 294, in _run_group
    refs = self.dvc.make_analyses(refs)
root                                    : 2018-04-16 11:44:14,666 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/dvc/dvc.py", line 739, in make_analyses
    branches = {ei: get_repository_branch(os.path.join(paths.repository_dataset_dir, ei)) for ei in exps}
root                                    : 2018-04-16 11:44:14,666 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/dvc/dvc.py", line 739, in <dictcomp>
    branches = {ei: get_repository_branch(os.path.join(paths.repository_dataset_dir, ei)) for ei in exps}
root                                    : 2018-04-16 11:44:14,666 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/git_archive/repo_manager.py", line 46, in get_repository_branch
    r = Repo(path)
root                                    : 2018-04-16 11:44:14,666 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/git/repo/base.py", line 126, in __init__
    raise NoSuchPathError(epath)
root                                    : 2018-04-16 11:44:14,666 CRITICAL  (MainThread) git.exc.NoSuchPathError: /Users/mcintosh/Pychron_dev/data/.dvc/repositories/Irradiation-NM-274
root                                    : 2018-04-16 11:44:14,666 CRITICAL  (MainThread) ============ End Exception ==========

wcmcintosh, 2018-04-16:
Am I correct that this error has do do with not fetching the Irradiation-NM-274 repository?

jirhiker, 2018-04-16:
close. Here are the relevant lines

```
git.cmd : 2018-04-16 11:44:14,624 DEBUG (MainThread) Popen(['git', 'merge', 'FETCH_HEAD'], cwd=/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Cascades, universal_newlines=False, shell=None)
GitRepoManager : 2018-04-16 11:44:14,662 WARNING (MainThread) Git command failed. Cmd('git') failed due to: exit code(128)
cmdline: git merge FETCH_HEAD
stderr: 'error: Merging is not possible because you have unmerged files.
hint: Fix them up in the work tree, and then use 'git add/rm '
hint: as appropriate to mark resolution and make a commit.
fatal: Exiting because of an unresolved conflict.', merge
```

Your Cascades repository could not be merged automatically because your local copy has changes that cannot be automatically resolved. Good example of why mixing analyses between repositories is bad practice

wcmcintosh, 2018-04-16:
So how do I fix this?

wcmcintosh, 2018-04-16:
After a pull of cascades repository I get this  <Note there grammar error>

You are 9 behind and 8 commits ahead. Their is potential for conflicts that you will have to resolve.Would you like to Continue?

wcmcintosh, 2018-04-16:
Note the grammar error in the error statement

closed 2018-04-16 by wcmcintosh

wcmcintosh, 2018-04-16:
Pycharm output from pull & push

pull gave feedback shown above
no Pychron feedback from push

I was expecting an opportunity to choose ours/vs theirs

ignore exception
urllib3.connectionpool                  : 2018-04-16 13:32:52,311 DEBUG     (MainThread) Starting new HTTPS connection (1): github.com
urllib3.connectionpool                  : 2018-04-16 13:32:52,652 DEBUG     (MainThread) https://github.com:443 "GET /NMGRL/pychron/issues/new HTTP/1.1" 302 None
urllib3.connectionpool                  : 2018-04-16 13:32:52,780 DEBUG     (MainThread) https://github.com:443 "GET /login?return_to=https%3A%2F%2Fgithub.com%2FNMGRL%2Fpychron%2Fissues%2Fnew HTTP/1.1" 200 None
libpng warning: iCCP: known incorrect sRGB profile
2018-04-16 13:35:08.290 python[16539:23544867] modalSession has been exited prematurely - check for a reentrant call to endModalSession:
git.cmd                                 : 2018-04-16 13:35:10,123 DEBUG     (MainThread) Popen(['git', 'log', 'master', '--pretty=%H|%cn|%ce|%ct|%s'], cwd=/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Cascades, universal_newlines=False, shell=None)
GitRepoManager                          : 2018-04-16 13:35:17,164 DEBUG     (MainThread) ahead behind
git.cmd                                 : 2018-04-16 13:35:17,165 DEBUG     (MainThread) Popen(['git', 'fetch', 'origin'], cwd=/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Cascades, universal_newlines=False, shell=None)
git.cmd                                 : 2018-04-16 13:35:17,794 DEBUG     (MainThread) Popen(['git', 'status', '-sb'], cwd=/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Cascades, universal_newlines=False, shell=None)
GitRepoManager                          : 2018-04-16 13:35:17,850 DEBUG     (MainThread) Smart pull ahead: 8 behind: 9
2018-04-16 13:35:24.946 python[16539:23544867] modalSession has been exited prematurely - check for a reentrant call to endModalSession:
git.cmd                                 : 2018-04-16 13:35:24,948 DEBUG     (MainThread) Popen(['git', 'rebase', '--preserve-merges', 'origin/master'], cwd=/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Cascades, universal_newlines=False, shell=None)
GitRepoManager                          : 2018-04-16 13:35:25,092 WARNING   (MainThread) Git command failed. Cmd('git') failed due to: exit code(1)
  cmdline: git rebase --preserve-merges origin/master
  stdout: 'a-01/baselines/-J-3831.base.json: needs merge
a-01/baselines/-J-3832.base.json: needs merge
a-01/baselines/-J-3833.base.json: needs merge
a-01/baselines/-J-3834.base.json: needs merge
a-01/baselines/-J-3835.base.json: needs merge
a-01/baselines/-J-3836.base.json: needs merge
a-01/baselines/-J-3841.base.json: needs merge
a-01/baselines/-J-3842.base.json: needs merge
a-01/baselines/-J-3843.base.json: needs merge
a-01/baselines/-J-3844.base.json: needs merge
a-01/baselines/-J-3845.base.json: needs merge
a-01/baselines/-J-3846.base.json: needs merge
a-01/baselines/-J-3974.base.json: needs merge
a-01/baselines/-J-3975.base.json: needs merge
a-01/intercepts/-J-3831.inte.json: needs merge
a-01/intercepts/-J-3832.inte.json: needs merge
a-01/intercepts/-J-3833.inte.json: needs merge
a-01/intercepts/-J-3834.inte.json: needs merge
a-01/intercepts/-J-3835.inte.json: needs merge
a-01/intercepts/-J-3836.inte.json: needs merge
a-01/intercepts/-J-3841.inte.json: needs merge
a-01/intercepts/-J-3842.inte.json: needs merge
a-01/intercepts/-J-3843.inte.json: needs merge
a-01/intercepts/-J-3844.inte.json: needs merge
a-01/intercepts/-J-3845.inte.json: needs merge
a-01/intercepts/-J-3846.inte.json: needs merge
a-01/intercepts/-J-3974.inte.json: needs merge
a-01/intercepts/-J-3975.inte.json: needs merge
ba-01/baselines/-J-1619.base.json: needs merge
ba-01/baselines/-J-1621.base.json: needs merge
ba-01/baselines/-J-1622.base.json: needs merge
ba-01/intercepts/-J-1619.inte.json: needs merge
ba-01/intercepts/-J-1621.inte.json: needs merge
ba-01/intercepts/-J-1622.inte.json: needs merge
bu-FC/baselines/-J-5422.base.json: needs merge
bu-FC/baselines/-J-5423.base.json: needs merge
bu-FC/baselines/-J-5424.base.json: needs merge
bu-FC/baselines/-J-5425.base.json: needs merge
bu-FC/baselines/-J-5426.base.json: needs merge
bu-FC/baselines/-J-5427.base.json: needs merge
bu-FC/baselines/-J-5428.base.json: needs merge
bu-FC/baselines/-J-5445.base.json: needs merge
bu-FC/baselines/-J-5446.base.json: needs merge
bu-FC/baselines/-J-5447.base.json: needs merge
bu-FC/baselines/-J-5448.base.json: needs merge
bu-FC/baselines/-J-5449.base.json: needs merge
bu-FC/baselines/-J-5450.base.json: needs merge
bu-FC/baselines/-J-6105.base.json: needs merge
[... 52 lines cut ...]
root                                    : 2018-04-16 13:35:25,369 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/git_archive/repo_manager.py", line 651, in smart_pull
    conflict_paths = [os.path.relpath(x, self.path) for x in out.splitlines()]
root                                    : 2018-04-16 13:35:25,369 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/git_archive/repo_manager.py", line 651, in <listcomp>
    conflict_paths = [os.path.relpath(x, self.path) for x in out.splitlines()]
root                                    : 2018-04-16 13:35:25,369 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/posixpath.py", line 460, in relpath
    genericpath._check_arg_types('relpath', path, start)
[... cut ...]
git.cmd                                 : 2018-04-16 13:35:45,593 DEBUG     (MainThread) Popen(['git', 'fetch', 'origin'], cwd=/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Cascades, universal_newlines=False, shell=None)
git.cmd                                 : 2018-04-16 13:35:46,017 DEBUG     (MainThread) Popen(['git', 'status', '-sb'], cwd=/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Cascades, universal_newlines=False, shell=None)
GitRepoManager                          : 2018-04-16 13:35:46,070 DEBUG     (MainThread) Smart pull ahead: 8 behind: 9