NMGRL/pychron #1374: This is a test

jirhiker, 2018-04-16:
active branch=develop

active analyses=

```
Traceback (most recent call last):
  File "//anaconda/envs/pychron3/lib/python3.5/site-packages/pyface/ui/qt4/action/action_item.py", line 346, in _qt4_on_triggered
    self.controller.perform(action, action_event)
  File "//anaconda/envs/pychron3/lib/python3.5/site-packages/pyface/tasks/action/task_action_controller.py", line 31, in perform
    return action.perform(event)
  File "//anaconda/envs/pychron3/lib/python3.5/site-packages/pyface/tasks/action/listening_action.py", line 58, in perform
    method()
  File "/Users/ross/Programming/github/pychron3/pychron/pipeline/tasks/task.py", line 198, in pipeline_recall
    print(a)
NameError: name 'a' is not defined

```

wcmcintosh, 2018-04-17:
Jake:  Not sure why this issue is not reproducing for you.  Matt and I just submitted an automatic bug report, with title, and it posted with "no title provided" (Issue 1379)

jirhiker, 2018-04-17:
There must be an issue with how you are setting the Title. "no title provided" is not something github is doing. If no title is entered pychron automatically substitutes in "no title provided". (this is because github will not accept an issue with no title)

closed 2018-05-22 by jirhiker