NMGRL/pychron #1376: Spectrum reseting age scale when changing spectrum preferences

mjz1983, 2018-04-17:
When I change the Y axis (apparent age) for the spectrum, then change the prefences (specifically adding Y error to K/Ca and 40Ar), the apparent age resets to the default.

mjz1983, 2018-04-17:
In fact, all Y axes (age, k/ca, and 40Ar) reset to default scales when changing preferences.

jirhiker, 2018-04-17:
what mechanism are you using to modify the limits. right click and using the editor or grabbing the axis and dragging?

mjz1983, 2018-04-17:
First, I right click on the scale then set the upper and lower limits. Then, when I click on "spectrum" in the pipeline details window, set the spectrum options, the scale changes back to the default upper and lower limits.

closed 2019-07-21 by jirhiker