NMGRL/pychron #1379: No Title Provided

nmgrluser, 2018-04-17:
active branch=release/py3/v18.2

active analyses=66373-01D

```
Traceback (most recent call last):
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.4/site-packages/pyface/ui/qt4/action/action_item.py", line 346, in _qt4_on_triggered
    self.controller.perform(action, action_event)
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.4/site-packages/pyface/tasks/action/task_action_controller.py", line 31, in perform
    return action.perform(event)
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.4/site-packages/pyface/tasks/action/listening_action.py", line 58, in perform
    method()
  File "/Users/argonlab2/.pychron.1/updates/pychron/pipeline/tasks/task.py", line 187, in diff_analysis
    editor.set_diff(left)
  File "/Users/argonlab2/.pychron.1/updates/pychron/pipeline/editors/diff_editor.py", line 243, in set_diff
    self._set_values(left, right, isotopes)
  File "/Users/argonlab2/.pychron.1/updates/pychron/pipeline/editors/diff_editor.py", line 300, in _set_values
    rvalue=nominal_value(right.kca) ** -1))
TypeError: unsupported operand type(s) for ** or pow(): 'NoneType' and 'int'

```

mjz1983, 2018-04-17:
Title was provided for this issue.

mjz1983, 2018-04-17:
Just to clarify, this is an automated bug not one submitted through help>add request/submit bug

wcmcintosh, 2018-04-17:
Seems to work fine now, ready to close

closed 2018-04-17 by jirhiker