NMGRL/pychron #1386: find flux monitor error

wcmcintosh, 2018-04-18:
error in first step of flux pipeline
occurs for NM-281A
NM-274E,F,G worked fine

root                                    : 2018-04-18 13:53:12,895 CRITICAL  (MainThread) ============ Exception ==============
root                                    : 2018-04-18 13:53:12,895 CRITICAL  (MainThread) Traceback (most recent call last):
root                                    : 2018-04-18 13:53:12,895 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/ui/qt4/action/action_item.py", line 346, in _qt4_on_triggered
    self.controller.perform(action, action_event)
root                                    : 2018-04-18 13:53:12,895 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/tasks/action/task_action_controller.py", line 31, in perform
    return action.perform(event)
root                                    : 2018-04-18 13:53:12,896 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/tasks/action/listening_action.py", line 58, in perform
    method()
root                                    : 2018-04-18 13:53:12,896 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 308, in run
    self._run_pipeline()
root                                    : 2018-04-18 13:53:12,896 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 489, in _run_pipeline
    self._run('run pipeline', 'run_pipeline')
root                                    : 2018-04-18 13:53:12,896 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 468, in _run
    if not getattr(self.engine, func)():
root                                    : 2018-04-18 13:53:12,896 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/engine.py", line 722, in run_pipeline
    node.run(state)
root                                    : 2018-04-18 13:53:12,896 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/nodes/find.py", line 129, in run
    monitors = self.dvc.make_analyses(ans)
root                                    : 2018-04-18 13:53:12,896 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/dvc/dvc.py", line 739, in make_analyses
    branches = {ei: get_repository_branch(os.path.join(paths.repository_dataset_dir, ei)) for ei in exps}
root                                    : 2018-04-18 13:53:12,896 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/dvc/dvc.py", line 739, in <dictcomp>
    branches = {ei: get_repository_branch(os.path.join(paths.repository_dataset_dir, ei)) for ei in exps}
root                                    : 2018-04-18 13:53:12,896 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/git_archive/repo_manager.py", line 47, in get_repository_branch
    b = r.active_branch
root                                    : 2018-04-18 13:53:12,896 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/git/repo/base.py", line 663, in active_branch
    return self.head.reference
root                                    : 2018-04-18 13:53:12,897 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/git/refs/symbolic.py", line 251, in _get_reference
    raise TypeError("%s is a detached symbolic reference as it points to %r" % (self, sha))
root                                    : 2018-04-18 13:53:12,897 CRITICAL  (MainThread) TypeError: HEAD is a detached symbolic reference as it points to 'daebc7aaf130f8d356ba4f3f3992b49926aea1fb'
root                                    : 2018-04-18 13:53:12,897 CRITICAL  (MainThread) ============ End Exception ==========

wcmcintosh, 2018-04-18:
Same error occurs if analyses chosen through browser

wcmcintosh, 2018-04-18:
Same error occurs for NM-281B

wcmcintosh, 2018-04-18:
Not getting consistent behavior with this error.  In some cases NM-274E works fine.  In other cases it triggers this error.  In some cases error occurs before sample list appears.  In other cases it happens after sample list appears.

jirhiker, 2018-04-19:
That error is local to your computer. It suggests there is a problem with one or more of your local repositories. 

PLEASE use three back ticks before and after the pasted error message. It makes it much easier to read

jirhiker, 2018-04-19:
you can try opening "Experiment Repositories" select NM-281 and delete your local changes using "Delete Local Changes"

wcmcintosh, 2018-04-20:
I tried deleting local changes in NM-281 and still get the same error

```
root                                    : 2018-04-19 23:56:53,139 CRITICAL  (MainThread) ============ Exception ==============
root                                    : 2018-04-19 23:56:53,139 CRITICAL  (MainThread) Traceback (most recent call last):
root                                    : 2018-04-19 23:56:53,139 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/ui/qt4/action/action_item.py", line 346, in _qt4_on_triggered
    self.controller.perform(action, action_event)
root                                    : 2018-04-19 23:56:53,139 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/tasks/action/task_action_controller.py", line 31, in perform
    return action.perform(event)
root                                    : 2018-04-19 23:56:53,140 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/tasks/action/listening_action.py", line 58, in perform
    method()
root                                    : 2018-04-19 23:56:53,140 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 308, in run
    self._run_pipeline()
root                                    : 2018-04-19 23:56:53,140 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 489, in _run_pipeline
    self._run('run pipeline', 'run_pipeline')
root                                    : 2018-04-19 23:56:53,140 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 468, in _run
    if not getattr(self.engine, func)():
root                                    : 2018-04-19 23:56:53,140 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/engine.py", line 722, in run_pipeline
    node.run(state)
root                                    : 2018-04-19 23:56:53,140 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/nodes/find.py", line 129, in run
    monitors = self.dvc.make_analyses(ans)
root                                    : 2018-04-19 23:56:53,140 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/dvc/dvc.py", line 739, in make_analyses
    branches = {ei: get_repository_branch(os.path.join(paths.repository_dataset_dir, ei)) for ei in exps}
root                                    : 2018-04-19 23:56:53,140 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/dvc/dvc.py", line 739, in <dictcomp>
    branches = {ei: get_repository_branch(os.path.join(paths.repository_dataset_dir, ei)) for ei in exps}
root                                    : 2018-04-19 23:56:53,140 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/git_archive/repo_manager.py", line 47, in get_repository_branch
    b = r.active_branch
root                                    : 2018-04-19 23:56:53,140 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/git/repo/base.py", line 663, in active_branch
    return self.head.reference
root                                    : 2018-04-19 23:56:53,141 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/git/refs/symbolic.py", line 251, in _get_reference
    raise TypeError("%s is a detached symbolic reference as it points to %r" % (self, sha))
root                                    : 2018-04-19 23:56:53,141 CRITICAL  (MainThread) TypeError: HEAD is a detached symbolic reference as it points to '5f26a8002477ddb37dd84d760890901f1d2d98e6'
root                                    : 2018-04-19 23:56:53,141 CRITICAL  (MainThread) ============ End Exception ==========

```

wcmcintosh, 2018-04-20:
can I delete the repository and start afresh?

jirhiker, 2018-04-20:
yes. Thats one of the beauties of repos. They are "cheap" to create/recreate

Delete Local changes would effectively do the same thing. Its not clear from the error that Irradiation-NM-281 is the culprit. So maybe Delete local changes worked but you just deleted local changes from the wrong repo. 

anyways I would do the following

```
# backup the existing repo
cd PychronDev/data/.dvc/repositories
mv ./Irradiation-NM-281 ./Irradiation-NM-281_backup

# Stand by for a GUI option. I will add Backup Repo to the Experiment Repositories toolbar
```

Relaunch pychron and when analyses from Irradiation-NM-281 are request the repository will be automatically cloned to your computer.

closed 2018-04-25 by jirhiker