NMGRL/pychron #1402: table group

jirhiker, 2018-04-30:
add a table group identifier to the tag file

add set table group tag in same places as set_tag.
use same set_tag code except no need to interact with database. only need to write/commit to tag file

0,0,0,1,1,1
a,b,c,d,e,f
...
s=IntermediateAnalysis(d,e,f)
a,b,c,s
 
the summary age is the sum of a,b,c,s NOT a,b,c,d,e,f

wcmcintosh, 2018-04-30:
save grouping does not seem to be working
no grouping remains after reset

wcmcintosh, 2018-04-30:
grouping process working OK now,
but now both individual analyses and sample-mean values multiplying by 1000x when set for ka

wcmcintosh, 2018-04-30:
problem with 40Ar column being hidden now fixed

wcmcintosh, 2018-04-30:
Save group working well
How about included a clear group as well?  Once  grouping is saved, there is apparently currently  no way to get rid of it

wcmcintosh, 2018-04-30:
Question on table sub-grouping:  Should subgrouping be contained within a dedicated node?  So for example, it could be used for ideograms as well as the table.  This would allow matching numbers in tables and figures.  I guess whether sub-grouping is provided in ideograms is a separate question from  whether it is in a dedicated node.

Sug-grouping is providing capabilities never available in the pre-pychron world, outside of cutting and pasting.  It is quite useful for data sets where a few crystals are first step heated, and subsequent crystals are run as two-steppers.

Your implementation of sub-grouping allows a lot of flexibility for non-simple data sets, which while not the rule, are not that uncommon.  Thanks for all the work on this.

jirhiker, 2018-05-01:
a simple solution may be to use the existing grouping node to group analyses by the "table_group" tag.

Subgrouping is a little tricky and I wasn't totally enamored with the idea at first, interpretated ages was supposed to cover this functionality. However each time I thought about my response to "How would you accomplish this otherwise" I was unable to come up with any acceptable alternative. 

I can also see subgrouping being very useful for DS ideograms and tables

wcmcintosh, 2018-05-23:
Table grouping now working perfectly as far as I can tell. 
The dream of a unified table format has been achieved.  
Time to close.

closed 2018-05-23 by jirhiker