NMGRL/pychron #1410: error on clear grouping in sub-group node

wcmcintosh, 2018-05-02:
active branch=None

active analyses=64102-01A,64102-01B,64102-01C,64102-02A,64102-02B,64102-02C,64102-03A,64102-03B,64102-03C,64102-04A,64102-04B,64102-04C,64102-05A,64102-05B,64102-05C,64102-06A,64102-06B,64102-06C,64102-07A,64102-07B,64102-07C,64102-08A,64102-08C,64102-09A,64102-09B,64102-09C,64102-10A,64102-10B,64102-10C,64102-11A,64102-11B,64102-11C,64102-12A,64102-12B,64102-12C

description=""

Traceback
```
Traceback (most recent call last):
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/ui/qt4/action/action_item.py", line 163, in _qt4_on_triggered
    self.controller.perform(action, action_event)
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traitsui/qt4/editor.py", line 269, in perform
    self.ui.do_undoable( self._perform, action )
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traitsui/ui.py", line 716, in do_undoable
    action( *args, **kw )
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traitsui/qt4/editor.py", line 291, in _perform
    method( info, selection )
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/editors/arar_table_editor.py", line 108, in clear_grouping
    obj.clear_grouping()
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/editors/arar_table_editor.py", line 131, in clear_grouping
    self._compress_groups()
AttributeError: 'ArArTableEditor' object has no attribute '_compress_groups'

```

wcmcintosh, 2018-05-02:
good title this time. I clicked in the description box after typing the title

jirhiker, 2018-05-02:
sounds like you need to either hit enter after typing in a title OR click on some other widget within the window

closed 2018-05-22 by jirhiker