NMGRL/pychron #1423: Production ratio names

nmgrluser, 2018-05-05:
When convenient can we clean up and normalize the PR names?

nmgrluser, 2018-05-05:
Jake, Thanks for that info.  It  seems like the local sets would be rarely used.
How can  I access and modify the global default set?
Also, the production data entry pane could use a save as like similar windows.

nmgrluser, 2018-05-05:
Can I add global default PR sets for a few additional irradiations?  Or should I just add local versions for the moment.

nmgrluser, 2018-05-05:
Wow.  In trying to match production ratios between the two programs, I keep finding problems with Mass Spec PR files, mostly where the error fields have been used to store various unrelated values, much as in the preferences for decay values, where the error fields aren't being used.  I expect that this has been injecting extra uncertainty into various mass spec calculations, certainly for K/Ca.  An interesting and wacky source or error.

jirhiker, 2018-05-06:
Don't think you understand wrt global and irradiation-specific PRs. 

Both are necessary. Well in actuality only the irradiation-specific PRs are necessary but the global ones serve as defaults. So when a new irradiation is entered the user does not need to manually populate the PR, she can select from the global set.

closed 2018-05-22 by jirhiker