NMGRL/pychron #1424: Error   fitting air ic factor to Felix data

nmgrluser, 2018-05-05:
active branch=release/py3/v18.2

active analyses=a-02-F-857,a-02-F-858,a-02-F-859,a-02-F-860,a-02-F-861,a-02-F-862,a-02-F-863,a-02-F-864,a-02-F-865,a-02-F-866,a-02-F-867,a-02-F-868,a-02-F-869,a-02-F-870,a-02-F-871,a-02-F-872,a-02-F-873,a-02-F-874,a-02-F-875,a-02-F-876,a-02-F-877,a-02-F-878,a-02-F-879,a-02-F-880,a-02-F-881,a-02-F-882,a-02-F-883,a-02-F-884,a-02-F-885,a-02-F-886

description=""

Traceback
```
Traceback (most recent call last):
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.4/site-packages/pyface/ui/qt4/action/action_item.py", line 346, in _qt4_on_triggered
    self.controller.perform(action, action_event)
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.4/site-packages/pyface/tasks/action/task_action_controller.py", line 31, in perform
    return action.perform(event)
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.4/site-packages/pyface/tasks/action/listening_action.py", line 58, in perform
    method()
  File "/Users/argonlab2/.pychron.1/updates/pychron/pipeline/tasks/task.py", line 308, in run
    self._run_pipeline()
  File "/Users/argonlab2/.pychron.1/updates/pychron/pipeline/tasks/task.py", line 489, in _run_pipeline
    self._run('run pipeline', 'run_pipeline')
  File "/Users/argonlab2/.pychron.1/updates/pychron/pipeline/tasks/task.py", line 468, in _run
    if not getattr(self.engine, func)():
  File "/Users/argonlab2/.pychron.1/updates/pychron/pipeline/engine.py", line 731, in run_pipeline
    node.run(state)
  File "/Users/argonlab2/.pychron.1/updates/pychron/pipeline/nodes/fit.py", line 87, in run
    if self.check_refit(unks):
  File "/Users/argonlab2/.pychron.1/updates/pychron/pipeline/nodes/fit.py", line 64, in check_refit
    if self._check_refit(ui):
  File "/Users/argonlab2/.pychron.1/updates/pychron/pipeline/nodes/fit.py", line 196, in _check_refit
    self.warning('Data for detector missing {}'.format(k))
AttributeError: 'FitICFactorNode' object has no attribute 'warning'

```

jirhiker, 2018-05-05:
addressed but this is a consequence of human error. You are trying to do DetA/DetB when the analyses where not measured with DetA and/or DetB. e.g For Felix the ic factors are calculated using H1/L2(CDD), For Jan its H1/CDD

nmgrluser, 2018-05-05:
We realized our error.  Perhaps this human error could be trapped and dealt with in various ways.  In general, human mixing or mis-matching of Jan and Felix data during data processing can cause a variety of problems.

nmgrluser, 2018-05-05:
User error.

closed 2018-05-05 by nmgrluser