NMGRL/pychron #1427: Table request and thoughts

nmgrluser, 2018-05-05:
1) Non-urgent request - table formatting could use a little more granularity, specifically separating number formatting digits for isotopes vs IC factors.  J already has dedicated formatting. Number of digits working correctly after yesterday's -1 correction.

2) Auto subgrouping is working quite well

Sending subgrouped data to a Step Heat format table does currently gives an error (#1425, etc.) issues but this can be easily avoided 

3) If subgroup mean-data could be made a little richer, I still think that the fusion and spectra tables could be seamlessly merged into a single format.  The subgroups could be treated more like groups in the current Step Heat table format.

Subgroup pull down could include additional choices for primary subgroup mean-data used in the calculations:
plateau
weighted mean
plateau else weighted mean
itegrated
isochron

Grouped data could  use the  primary requested  subgroup mean-data (e.g plateau) age, and still display  secondary subgroup mean-data (e.g weighted-mean integrated, isochron),with these lines marked as not included in mean (unless you think this makes the table too messy) 

Steps within groups could be marked as not included in the mean (first column) for graphic clarity.

Subgroup plateau line could include additional info (e.g. %39Ar on plateau, "no plateau" rather than zeros where appropriate, clarify which steps used in plateau)

If Pycron handled step-heating data this way, the need for two-different formats might just melt away.  Sample-mean values would only be calculated when needed (e.g. multiple aliquots), and aliquot-mean values would serve for single-aliquot samples.

Does this make sense to you?

4) I realized the problem with the crazy weighted mean K/Ca values yesterday.  The irradiation files on both Mass Spec and Pychron had Ca/K values set to 1.96±1.96, instead of 196±0 which added 100% error to the K/Ca values, greatly de-weighting larger K/Ca values in the weighted-mean calculations.

5) I'll work on tuning up the footnotes.

jirhiker, 2018-05-05:
This only makes partial sense. what doesn't make sense is the same idea i keep coming back to, "explicit is better than implicit". You need (I want you) to grok that.

The need for multiple table formats is absolutely necessary. (Not all users are going to want ArAr tables).  Explicitly defining what type of table you are making is much better than pychron trying to determine the table type implicitly from the data. 

In every case Ive taken a short cut using implicit definitions it has come back to bit me. 
Explicit is better then implicit

Having more info in the subgroup lines is fine. Making Spectrum and Fusion tables more unified is fine, Pychron intuiting what the user intends is not fine

nmgrluser, 2018-05-05:
In this case, it is not so much intuiting as simply accommodating different types of data within a single-format.  I believe that what currently appears in a Step-heat format table can covered by our current sub-groups.  All we are doing in the Step-heat format table is basically grouping by aliquot.

On another front, see issue

nmgrluser, 2018-05-05:
Separate table question.  When subgrouping, the user has the opportunity to auto or manually subgroup, then choose plot options.  But table options are presented when the pipeline is shown as being in the final save node. I believe that you explained why, but I still don't understand.  Wouldn't most pipelines be in the Analysis Table node at this point?

nmgrluser, 2018-05-05:
One more subgroup observation:  In a data set with x fusion analyses followed by sub-groups, the sub-group is numbered weighted_mean_(x+1) rather than weighted_mean_0.  This does not cause any problem at all with the table sub-groups.  Perhaps you mean it to work this way - just an observation.

nmgrluser, 2018-05-05:
Weighted-mean K/Ca values now looking much more normal without the extra 100% error.

nmgrluser, 2018-05-05:
Table IC factor questions:  
Is disc ± error necessary?
Why the current order: AX, CDD,  H2, L1, H1, L2?  
Does ICcdd cover CDD for both Jan and Felix (no problem if it does, just checking)?
Should table include system (Felix vs Jan) so that notes could clarify isotope/detector relationship for Jan vs Felix? Jan vs Felix  could appear in the header of the human table.

nmgrluser, 2018-05-05:
Do machine and human include irradiation level? I see it in the summary table.

jirhiker, 2018-05-06:
I still don't think your grasping "explicit vs implicit". Its better to tell it to do something then for it to have to infer to do something based on some criteria. Your mistake always is that you want the computer to make decisions. Computers only do what they are told. So it has to be configured. and more configuration means more up front user interaction and more work to provide the required interfaces.  In other words the processes are algorithmic and its very difficult to accommodate all the permutations. And its much too limiting. 

It was the same thing with the skip tag. How the skip tag is applied is explicitly defined in the preferences. Its not inferred from its name or the time was created any attribute of the tag, its explicit

Making a step heat table because the analyses have an step label is the implicit aspect and what im saying is the general problem. 

Subgroups are not for step heating data. they don't make logical sense. 

One thing to keep in mind is the the nodes are supposed to be elementary building blocks. The way to accomplish the complex behavior you are looking for is not to alter the building blocks but to string them together. This is the purpose of user defined pipelines.  For example if you want to make a step heat table you use the step heat table pipeline, or if you want a subgrouped ds table use your custom subgrouped ds table. My point! Don't have a single pipeline that needs to make discussions about what to do based on the input. Be explicit.

jirhiker, 2018-05-06:
"Separate table question. When subgrouping, the user has the opportunity to auto or manually subgroup, then choose plot options. But table options are presented when the pipeline is shown as being in the final save node. I believe that you explained why, but I still don't understand. Wouldn't most pipelines be in the Analysis Table node at this point?"

Think about it this way. You are presented with Plot Options before you "make" the figure. You are presented with Table Options before you "make" the table.  In the case of plots "make" means view in pychron. In the case of tables "make" means save as an excel file.

closed 2018-05-14 by jirhiker