NMGRL/pychron #1429: Questions

nmgrluser, 2018-05-06:
When viewing repositories, did we used to have a status color coding or is there a better way to check if all changes pushed?

jirhiker, 2018-05-06:
repositories are colored red if they have changes... but the repositories are not automatically checked for changes. 

When the Experiment Repository window is opened you are asked to "Check all repositories for changes"
Depending on the number of repos on your computer this can be time consuming. 

Once the Experiment Repository window is open you can select individual (or a group?) of repositories then select "Find Changes" from the toolbar

wcmcintosh, 2018-05-06:
Is Slack preferred for questions? I should add it to Argonlab2.

closed 2018-05-06 by wcmcintosh