NMGRL/pychron #1438: Monday table notes and questions

wcmcintosh, 2018-05-07:
Table making is working quite fantastically
I love plateau else mean

Minor bugs:
No Plateau needs bottom borders

I have a few final requests for finishing touches
I am trying for the single-format holy grail, not user mind-reading

1) Add n=xx to plateau ages
2) Add %39 next to K20, only populate for multi-aliquot steps, subgroup plateau ages, and subgroup  mean ages, otherwise leave cell empty, make column invisible if no cells are populated
3) Populate %K2O for appropriate single step analyses or subgroups, make column invisible if no cells are populated

i'll add more points on this later

jirhiker, 2018-05-07:
2. What is a multi-aliquot step? what do you mean by %39.  Do you mean on the subgroup summary line display the %39 the plateau represents of the total spectrum. if so for the subgroup mean age does %39 = 39_non_omitted_analyses/39_all_analyses *100 ?

3. k2o is already displayed but since no weight associated with analysis no way to calc k2o

wcmcintosh, 2018-05-07:
#2.  Multi-aliquot step should read multi-step  aliquot.

%39 = 39_non_omitted_analyses/39_all_analyses *100
This is correct for %39 of a sub-group mean.  %39 is also traditionally shown for individual step  elements of a subgroup.%39 = 39_step/39_all_analyses *100
For analyses that are not part of a subgroup, leave %39 blank

#3 correct - but how about leaving the entry blank in these cases where there is no weight and therefore no K2O, rather than entering a zero.

wcmcintosh, 2018-05-07:
#4 Could one tag analyses from the sub-group review pane, or is this not readily accomplished?

#1 Add n=xx to plateau ages - this could be accomplished in the subgroups in the same was as it is shown in the sample weighted-mean data lines, e.g. n=13/15

wcmcintosh, 2018-05-07:
#5 Could interpreted ages be set during summary table making, if specially requested by the user?  This would allow users to work with the tables iteratively to winnow data, as an alternative or supplement to doing the same process with figures.  
Low priority request

closed 2018-05-14 by jirhiker