NMGRL/pychron #1442: slight subgrouping challenge

wcmcintosh, 2018-05-07:
In general, subgrouping is working perfectly.  Specifically, no subgroups are identified where an aliquot has only one valid analysis.

However, in a few cases where many analyses are omitted, sub grouping gets slightly confused

The attached file is a 3 sample excel table, subgrouped by aliquot
In the first sample, #5 is subgrouped where sub-grouping should have made no subgroups
The second sample is perfect
In the third sample, #5C is subgrouped where sub-grouping should have made no subgroups

This is an extreme example, where subgrouping was requested but none existed, pushing the limits of the algorithm

jirhiker, 2018-05-08:
no attached file

wcmcintosh, 2018-05-08:
[test1.xlsx](https://github.com/NMGRL/pychron/files/1985940/test1.xlsx)

wcmcintosh, 2018-05-08:
current table

[test2.xlsx](https://github.com/NMGRL/pychron/files/1985956/test2.xlsx)

wcmcintosh, 2018-05-09:
Current table pipeline grouping perfectly, and N=n1/n2 and %39 look good.  

"Plateau else mean" seems to be correctly using mean for non-plateau samples

However, Plateau calculations currently using all steps on plateau.  See e.g. 65761-01 where step A is not part of plateau.(Spectrum pipeline picks plateau correctly)

Also, could "plateau"/"wt. mean" and "n=" data  shift one column to the right, and current "plateau"/"wt. mean" column display the aliquot number, or the last aliquot number should more than one aliquot be included in the subgroup.  

This aliquot number for the subgroup would also be useful for numbering subgroup-mean values on an ideogram.

Lastly, can you think of a good way to indicate steps included in or excluded from the plateau?

closed 2018-05-14 by jirhiker