NMGRL/pychron #1447: Age spectrum not displaying label for lower limit on auxiliary plots

mjz1983, 2018-05-09:
For both the K/Ca (log scale) and radiogenic yield (linear scale), the lower limit label is not displayed. This would be an space issue if labels are on the same side of the spectra, but if one is plotted on the left and another on the right, the space issue is not an issue anymore. Should this be an option. See screen shot.
[image]

jirhiker, 2018-05-09:
it is a minor detail that in the current setup I do not have control over. Ive already spent a disproportionate time on log axes and labeling. 

There are other minor log axes issues that remain; maybe if I decide to fix them then this problem will also get fixed

jirhiker, 2018-05-10:
Hopefully when you see the new feature im working on now you will (momentarily) forget about formatting/display issues.

closed 2019-09-11 by jirhiker