NMGRL/pychron #1448: Missing material label in auto generated Title

mjz1983, 2018-05-09:
When I select auto generate title "Sample (identifier - aliquot, material)", the material label is missing from the title. I double-checked that the material information is present in LabNumber feature of pychron.  Tested on 15L-50, NM294 M3.
[image]

mjz1983, 2018-05-09:
I typed it in wrong. There is a space between the aliquot and the comma. I'm using a format from the drop down window. See screenshot.
[image]

jirhiker, 2018-05-09:
analyses where not saved correctly and do not contain material information. Fixing it now

jirhiker, 2018-05-09:
LipmanDulceDikes repository is now fixed so that the sample info saved with the analysis is the same thats in the database. 

Seems like when these analyses were run the sample info was not synced correctly and so was not saved with the run.  Probably not human error.

closed 2018-05-11 by jirhiker