NMGRL/pychron #1451: Question about dropped point in evolution

wcmcintosh, 2018-05-10:
See attached PDF of diff comparison, 

Pychron filter is set to off , evolutions do not show dropped points, but diff table does show dropped points.

Which is right? Are numbers of dropped points explicitly stated in recall tab? 

[9.pdf](https://github.com/NMGRL/pychron/files/1990275/9.pdf)

wcmcintosh, 2018-05-10:
Or are these the number of points that would have been dropped if the filter was on?

jirhiker, 2018-05-10:
seems like a bug. but I cant see why fN==N when Filter=no looking at the code

wcmcintosh, 2018-05-10:
It looks like no points are actually being dropped, based agreement of the 40Ar intercept between the two programs, once the linear fit in MS actually happened.  There was some MS crashing along the way that caused trouble.

closed 2018-05-10 by jirhiker