NMGRL/pychron #1452: diff view 36Ar discrepancy

wcmcintosh, 2018-05-10:
The attached diff view shows 36 total for 25515-01 as 0.01062060 for Pychron
and 0.01064318 for MS

However the Ar36 BS corrected are the same for both analyses, and the 36Ar blanks are the same, and the 36 IC factors are the same (set to 1.005 ± 0 for both)

displaying the data in MS gives 36Ar = 0.010621, the correct value

0.010621 = (36Ar BS corrected - 36Ar blank) * IC

So how does Pychron retrieve 0.01064318 from the Mass Spec database?
Where is this number coming from?

[10.pdf](https://github.com/NMGRL/pychron/files/1990399/10.pdf)

wcmcintosh, 2018-05-10:
I set IC factors to zero, and still see a 36 difference in diff view, although the values are identical in MS and Py recall

MS and Py recall 36Ar=0.01056776
MS in diff view 36Ar=0.0106060   (where is this value coming from??)

[11.pdf](https://github.com/NMGRL/pychron/files/1991705/11.pdf)

wcmcintosh, 2018-05-10:
One other item which is not helping us in comparing MS and Py is the small lambda K difference
4.463 vs 4.464.  What can we do to make these match?  How about adding significant figures and minimizing rounding?

jirhiker, 2018-05-10:
Pychron doesn't do any rounding. It adds the decay constants for each branch together to get the total.

in pychron preferences set one decay branch to 0 and the other to the value in mass spec.

jirhiker, 2018-05-10:
it would appear that the diff view is not accessing the same data mass spec is during recall. 

Or they are accessing the same data but mass spec does something on the fly that alters it.

wcmcintosh, 2018-05-10:
OK, I get it now.  I wasn't saving the MS changes to the MS database.  Viewing the data with recall doesn't save the info the the database, even though an age is being calculated.  The user needs to manually save the analysis data. 
  I need to clarify what operations will save the MS data to the database.

wcmcintosh, 2018-05-10:
Question:  You mentioned that age calculations trigger saving analyses to the database.
Recall is apparently an exception to this, requiring a specific save request.
Do you know is all other age calculation operations automatically save the data in MS?

jirhiker, 2018-05-10:
I can't offer any definitive answer to that question. 

My guess is yes, other data reduction steps are automatically saved. How flux comes into play tho im not sure. For example if you recalculate the flux for a position, I dont know if  (kinda doubt it) mass spec than gets all analyses run from that position, recalculates an age then saves to database.

wcmcintosh, 2018-05-10:
Here the current diff, down to the dregs.

Only problems are eliminated points (not correct) and IC error comparisons (freaking out about zero)

Should we fix the 37ArK problem in mass spec?  Hasn't been a little while since we have compiled a new version?  Don't you miss that little bit of Al in your life?  That's a lowercase L, not a capitol I

[1.jpeg.pdf](https://github.com/NMGRL/pychron/files/1992421/1.jpeg.pdf)

jirhiker, 2018-05-10:
All indicators suggest that filtering is still turned on for 25515-01

1. Fits are suffixed with *
2. Diff view displays Ar40 Filter=yes

jirhiker, 2018-05-10:
we should not try to fix the 37ArK problem in mass spec. We would not be able to compile the source code we have for our current OSes

closed 2018-05-10 by jirhiker