NMGRL/pychron #1453: error arecalling blank using right click from found references pane

wcmcintosh, 2018-05-10:
active branch=None

active analyses=c-02-F-1249,c-02-F-1250,bu-FD-F-653,65766-08A,65766-08B,65766-08C,65766-08D,65766-08E

description=""

Traceback
```
Traceback (most recent call last):
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/ui/qt4/action/action_item.py", line 179, in _qt4_on_triggered
    action.perform(action_event)
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/action/action.py", line 104, in perform
    self.on_perform()
  File "/Users/mcintosh/Programming/git/pychron/pychron/graph/tools/analysis_inspector.py", line 83, in _recall_analysis
    ai.trigger_recall()
  File "/Users/mcintosh/Programming/git/pychron/pychron/database/records/isotope_record.py", line 120, in __getattr__
    return getattr(self.dbrecord, item)
AttributeError: 'AnalysisTbl' object has no attribute 'trigger_recall'

```

jirhiker, 2018-05-10:
What is the "found references pane"

jirhiker, 2018-05-10:
Pane has a very specific meaning in pychron and that error could not be coming from a "true" Pane

That error is triggered when right clicking a data point in a graph. Only a few panes in all of pychron display a graph and thats in data collection.

jirhiker, 2018-05-10:
something like Graphical Filter. 

Most windows have title bars that typically give a name or purpose, e.g. Ideogram Options, Edit Level, etc...

wcmcintosh, 2018-05-26:
Works fine now. Ready to close.

closed 2018-05-26 by jirhiker