NMGRL/pychron #1468: Error trying to branch

wcmcintosh, 2018-05-12:
Same error I think - mystery error

```
root                                    : 2018-05-11 18:00:41,328 CRITICAL  (MainThread) ============ Exception ==============
root                                    : 2018-05-11 18:00:41,328 CRITICAL  (MainThread) Traceback (most recent call last):
root                                    : 2018-05-11 18:00:41,328 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/ui/qt4/action/action_item.py", line 346, in _qt4_on_triggered
    self.controller.perform(action, action_event)
root                                    : 2018-05-11 18:00:41,328 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/tasks/action/task_action_controller.py", line 31, in perform
    return action.perform(event)
root                                    : 2018-05-11 18:00:41,328 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/tasks/action/listening_action.py", line 58, in perform
    method()
root                                    : 2018-05-11 18:00:41,328 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/dvc/tasks/repo_task.py", line 188, in add_branch
    if self._repo.create_branch(commit=commit.hexsha if commit else 'HEAD'):
root                                    : 2018-05-11 18:00:41,328 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/git_archive/repo_manager.py", line 537, in create_branch
    print(repo.branches, type(repo.branches))
root                                    : 2018-05-11 18:00:41,329 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/git/repo/base.py", line 242, in heads
    return Head.list_items(self)
root                                    : 2018-05-11 18:00:41,329 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/git/util.py", line 932, in list_items
    out_list.extend(cls.iter_items(repo, *args, **kwargs))
root                                    : 2018-05-11 18:00:41,329 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/git/refs/symbolic.py", line 592, in _iter_items
    for sha, rela_path in cls._iter_packed_refs(repo):  # @UnusedVariable
root                                    : 2018-05-11 18:00:41,329 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/git/refs/symbolic.py", line 92, in _iter_packed_refs
    raise TypeError("PackingType of packed-Refs not understood: %r" % line)
root                                    : 2018-05-11 18:00:41,329 CRITICAL  (MainThread) TypeError: PackingType of packed-Refs not understood: '# pack-refs with: peeled fully-peeled sorted'
root                                    : 2018-05-11 18:00:41,329 CRITICAL  (MainThread) ============ End Exception ==========
ignore exception

```

wcmcintosh, 2018-05-23:
Branching is operating without error, but I am not positive that checking out a branch is in fact switching branches.  For example, I branched IR998 to test, made changes in the history, pushed my changes, checked out the test branch, and ai do not see the changes.  The problem may lie in my understanding.

jirhiker, 2018-05-23:
this is the workflow

1. add branch or select existing
2. checkout branch
3. make changes

wcmcintosh, 2018-05-23:
I may have to show you this:
I add branch IR998:test
checkout test branch
delete youngest commit (perhaps this is the problem, am I not really deleting this commit?)
synch test branch
checkout test branch
then I see the change (latest commit not shown)

if I then checkout master branch then checkout test branch
the change is gone, and test = master (latest commit is back)

I must be missing something here
Could pychron more clearly indicate which branch is active on the local computer?

jirhiker, 2018-05-23:
The workflow makes it clear there is some misunderstanding.

"I add branch IR998:test " Ok this makes sense but its really IR998/local/test
"checkout test branch" Ok
"delete youngest commit (perhaps this is the problem, am I not really deleting this commit?)" You just created a branch that has no local changes so theres nothing to delete
"synch test branch" This is premature and not logical to do at this point. the branch test does not exist at `origin` (so nothing to pull) and you have no commits to share (so nothing to push)
"checkout test branch" you are already on test branch
"then I see the change (latest commit not shown)" ? dont follow

wcmcintosh, 2018-05-23:
My misunderstanding then.  When you display the list of commits for a given up-to-date branch (no local changes) , you can't just delete displayed commits to return to some earlier version?

jirhiker, 2018-05-23:
No. That is an atypical way to go back in time in git.

When you create a branch you specify the commit (HEAD the last commit is used if none selected) you want to branch at. 

"delete displayed commits to return to some earlier version" this is ok when the commits are local and have not been shared. But is very dangerous to use when changes have been shared. 

When changes have been shared this is the model to use. 

here is a master branch with 4 commits
M0--M1---M2---M3(HEAD)

if I want to rollback to M1 I branch from M1.
           
          /---------------------B0(HEAD)
M0--M1--M2---M3

wcmcintosh, 2018-05-23:
Ok, I added commits, and these are staying added , so it looks like branching is working well.  Time to close issue.

closed 2018-05-23 by jirhiker