NMGRL/pychron #1469: error creating analysis group

wcmcintosh, 2018-05-12:
```
SampleBrowserModel                      : 2018-05-11 18:06:51,800 DEBUG     (MainThread) load associated analysis groups for ['IR998 (Streck, M)']
root                                    : 2018-05-11 18:07:47,161 CRITICAL  (MainThread) ============ Exception ==============
root                                    : 2018-05-11 18:07:47,170 CRITICAL  (MainThread) Traceback (most recent call last):
root                                    : 2018-05-11 18:07:47,170 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traits/trait_notifiers.py", line 340, in __call__
    self.handler( *args )
root                                    : 2018-05-11 18:07:47,171 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/envisage/browser/sample_browser_model.py", line 220, in _add_analysis_group_button_fired
    project = next((p for p in self.oprojects if p.name == project and p.principal_investigator == pp))
root                                    : 2018-05-11 18:07:47,171 CRITICAL  (MainThread) StopIteration
root                                    : 2018-05-11 18:07:47,171 CRITICAL  (MainThread) ============ End Exception ==========

```

wcmcintosh, 2018-05-15:
still not making analysis successfully. no errors, following is pycharm output

```
pychronmeta                             : 2018-05-14 21:48:20,871 DEBUG     (MainThread) SELECT "ProjectTbl".name, "ProjectTbl".id, "ProjectTbl"."principal_investigatorID", "ProjectTbl".checkin_date, "ProjectTbl".comment, "ProjectTbl".lab_contact, "ProjectTbl".institution 
FROM "ProjectTbl" ORDER BY "ProjectTbl".name ASC
2018-05-14 21:48:47.239 python[45666:6350325] modalSession has been exited prematurely - check for a reentrant call to endModalSession:
pychronmeta                             : 2018-05-14 21:48:47,430 DEBUG     (MainThread) multiples row found for ProjectTbl ('name',) ('IR998',). Trying to get last row

```

jirhiker, 2018-05-15:
The analysis group was added to project IR998. Problem is that there are multiple IR998's and a projects name is not a unique identifier

jirhiker, 2018-05-15:
removed duplicate IR998 project

closed 2018-05-15 by jirhiker