NMGRL/pychron #1473: tables and figure pipelines near perfect

wcmcintosh, 2018-05-14:
tables and figures are really getting there. Following are a few more suggestions:

1) Tables.  mean ages for samples with single aliquots with a subgroup mean could use special treatment, whereby the normal weighted mean age is replaced by the single plateau or weighted-mean age and error of the subgroup, including appropriate n, mswd, etc.  In this way the single table format would accommodate the not-infrequent cases where only one age spectrum is collected per sample.  See the final sample in the attached example.
[Streck 2018.xlsx](https://github.com/NMGRL/pychron/files/2001179/Streck.2018.xlsx)

2) Figures.  The group by subgroup option is not quite doing the same thing as the tables.  The subgroups are grouped, but the individual steps are plotted rather than the mean or plateau ages of the subgroups. These points would require appropriate labeling if labelling is requested, perhaps 01p or 01m for plateaus or means.   Perhaps the plotting pipelines should include a subgroup node mirroring that of the tables? Having a plateau versus weighted-mean option for spectra figures would also mirror the tables. Sub-grouping could be accomplished either in the table or figure pipelines, then used by subsequent pipeline processes where appropriate.

The whole goal here is to have a tagged and sub-grouped data set produce identical figures and tables.

wcmcintosh, 2018-05-14:
I tried meanify to see if this was required to plot subgroups as their mean, but I don't think that this is the correct use of meanify. Plus I got an error.

jirhiker, 2018-05-14:
obvious a different name is required for "meanify".  

Age subgroupify?  (Treat subgroup as single age)

jirhiker, 2018-05-14:
1. concept does not fit well with current algorithm. would require redesign and extensive testing. 
Simply making 65773 as a separate step heat table gives you want you want.

wcmcintosh, 2018-05-15:
I think that data sets with multiple samples, some with multi aliquot data and some with single aliquot data, are not infrequent in NMGRL projects.  This is just the subset of "weighted-mean"  age for a sample in cases where n=1.  How does that violate the current algorithm?  Why does that require extensive testing and redesign?
 All of the sample ages, whether a single step-heat, multiple step heats, fusions, or whatever will then feed into the summary table without cutting and pasting.

jirhiker, 2018-05-15:
Fundamentally the problem is that your solution requires extensive testing and im really fed up with testing table making. 

These statements of "near perfect" and "almost complete" are irrelevant and disheartening. Every feature is infinitely far from being perfect or complete.

jirhiker, 2018-05-15:
I don't think you are adequately imagining all the permutations and variations that can occur. 

Take for example making a table of a single step heat sample.  Since only one aliquot is present no subgroup is applied, (even if the user selects "plateau" for the subgrouping) So now how does pychron know what type of summary age to display.  The problems only get compounded when mixing true step heat and fusion analyses.

wcmcintosh, 2018-05-21:
Issue overtaken by subsequent changes in program architecture.  Ready to close.

closed 2018-05-21 by jirhiker