NMGRL/pychron #1478: Add units for various parameters in recall window

mjz1983, 2018-05-18:
Currently there are no units for 

Lambda K - should be /yr or yr^-1

or 

Isotope intensities  - should be fA (perhaps add to column headers - I'm thinking about how to help new students with this comment)

jirhiker, 2018-05-18:
In reality the isotope intensities are unitless. What I mean is that pychron does not associate units with the intensities. So it would be misleading to display fA on the column header. The values are not necessarily fA either. for example the NGX does not convert the ion counter intensity to fA (like thermo does) so the intensity for the ion counter is read in cps and the ic factor is used to scale it to the faraday detectors. For students i think the more important thing to understand is the relative magnitude of the different isotopes. Its really only when youre trying to compare different instruments that units become important.

jirhiker, 2018-05-18:
I will add the units for the decay constants

closed 2018-05-20 by jirhiker