NMGRL/pychron #1486: Error running standard ideogram pipeline

wcmcintosh, 2018-05-21:
active branch=None

active analyses=65761-03,65761-04,65761-05,65761-06,65761-07,65761-08,65761-09,65761-10,65761-11,65761-12,65761-13,65761-14,65761-15

description=""

Traceback
```
Traceback (most recent call last):
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traits/trait_notifiers.py", line 340, in __call__
    self.handler( *args )
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/plot/figure_container.py", line 64, in _model_changed
    self.model_changed(True)
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/plot/figure_container.py", line 54, in model_changed
    self.model.refresh_panels()
  File "/Users/mcintosh/Programming/git/pychron/pychron/core/codetools/inspection.py", line 45, in dec
    return func(*args, **kw)
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/plot/models/figure_model.py", line 69, in refresh_panels
    ps = self._make_panels()
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/plot/models/figure_model.py", line 94, in _make_panels
    gi.make_figures()
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/plot/panels/figure_panel.py", line 53, in make_figures
    self.figures = self._make_figures()
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/plot/panels/ideogram_panel.py", line 51, in _make_figures
    if nsubgroups > 1 or bool(sg[0]):
TypeError: 'set' object does not support indexing

```

wcmcintosh, 2018-05-21:
Tested. same error

```
Traceback (most recent call last):
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traits/trait_notifiers.py", line 340, in __call__
    self.handler( *args )
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/plot/figure_container.py", line 64, in _model_changed
    self.model_changed(True)
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/plot/figure_container.py", line 54, in model_changed
    self.model.refresh_panels()
  File "/Users/mcintosh/Programming/git/pychron/pychron/core/codetools/inspection.py", line 45, in dec
    return func(*args, **kw)
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/plot/models/figure_model.py", line 69, in refresh_panels
    ps = self._make_panels()
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/plot/models/figure_model.py", line 94, in _make_panels
    gi.make_figures()
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/plot/panels/figure_panel.py", line 53, in make_figures
    self.figures = self._make_figures()
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/plot/panels/ideogram_panel.py", line 51, in _make_figures
    if nsubgroups > 1 or bool(sg[0]):
TypeError: 'set' object does not support indexing
```

jirhiker, 2018-05-21:
must not have updated code. this issue is addressed

wcmcintosh, 2018-05-21:
OK now.

closed 2018-05-21 by jirhiker