NMGRL/pychron #1490: error on set interpreted age

wcmcintosh, 2018-05-22:
error setting interpreted age for 65761-01

```
2018-05-22 13:46:08.076 python[56287:8021730] modalSession has been exited prematurely - check for a reentrant call to endModalSession:
PipelineEngine                          : 2018-05-22 13:46:08,079 DEBUG     (MainThread) Pre run failed Interpreted Ages<InterpretedAgeNode>
Pipeline Data Processing                : 2018-05-22 13:46:08,079 DEBUG     (MainThread) true run pipeline run_pipeline
Pipeline Data Processing                : 2018-05-22 13:46:08,079 DEBUG     (MainThread) run pipeline finished
Inspection                              : 2018-05-22 13:46:10,563 DEBUG     (MainThread) analyses called by get_interpreted_ages. parent call=<pychron.pipeline.plot.editors.spectrum_editor.SpectrumEditor object at 0x1c40473fc0> unks = sorted(self.analyses, key=key)
root                                    : 2018-05-22 13:46:10,619 CRITICAL  (MainThread) ============ Exception ==============
root                                    : 2018-05-22 13:46:10,619 CRITICAL  (MainThread) Traceback (most recent call last):
root                                    : 2018-05-22 13:46:10,620 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/ui/qt4/action/action_item.py", line 346, in _qt4_on_triggered
    self.controller.perform(action, action_event)
root                                    : 2018-05-22 13:46:10,620 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/tasks/action/task_action_controller.py", line 31, in perform
    return action.perform(event)
root                                    : 2018-05-22 13:46:10,620 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/tasks/action/listening_action.py", line 58, in perform
    method()
root                                    : 2018-05-22 13:46:10,620 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 324, in set_interpreted_age
    set_interpreted_age(self.dvc, ias)
root                                    : 2018-05-22 13:46:10,620 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/interpreted_age_factory.py", line 95, in set_interpreted_age
    info = iaf.edit_traits()
root                                    : 2018-05-22 13:46:10,620 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traitsui/handler.py", line 446, in edit_traits
    self.trait_view_elements(), handler, id, scrollable, args )
root                                    : 2018-05-22 13:46:10,620 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traitsui/view.py", line 433, in ui
    ui.ui( parent, kind )
root                                    : 2018-05-22 13:46:10,620 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traitsui/ui.py", line 222, in ui
    self.rebuild( self, parent )
root                                    : 2018-05-22 13:46:10,621 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traitsui/qt4/toolkit.py", line 148, in ui_livemodal
    ui_live.ui_livemodal( ui, parent )
root                                    : 2018-05-22 13:46:10,621 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traitsui/qt4/ui_live.py", line 44, in ui_livemodal
    _ui_dialog(ui, parent, BaseDialog.MODAL)
root                                    : 2018-05-22 13:46:10,621 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traitsui/qt4/ui_live.py", line 59, in _ui_dialog
    BaseDialog.display_ui(ui, parent, style)
root                                    : 2018-05-22 13:46:10,621 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traitsui/qt4/ui_base.py", line 482, in display_ui
    ui.owner.init(ui, parent, style)
root                                    : 2018-05-22 13:46:10,621 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traitsui/qt4/ui_live.py", line 177, in init
    self.add_contents(panel(ui), bbox)
root                                    : 2018-05-22 13:46:10,621 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traitsui/qt4/ui_panel.py", line 255, in panel
    panel = _GroupPanel(content[0], ui).control
root                                    : 2018-05-22 13:46:10,622 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traitsui/qt4/ui_panel.py", line 592, in __init__
    layout = self._add_items(content, inner)
root                                    : 2018-05-22 13:46:10,622 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traitsui/qt4/ui_panel.py", line 860, in _add_items
    editor.prepare(inner)
root                                    : 2018-05-22 13:46:10,622 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traitsui/editor.py", line 156, in prepare
    self.init( parent )
root                                    : 2018-05-22 13:46:10,622 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traitsui/qt4/table_editor.py", line 237, in init
    handler = factory.edit_view_handler))
root                                    : 2018-05-22 13:46:10,622 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traits/has_traits.py", line 1871, in edit_traits
    handler, id, scrollable, args )
root                                    : 2018-05-22 13:46:10,622 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traitsui/view.py", line 433, in ui
    ui.ui( parent, kind )
root                                    : 2018-05-22 13:46:10,622 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traitsui/ui.py", line 222, in ui
    self.rebuild( self, parent )
root                                    : 2018-05-22 13:46:10,622 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traitsui/qt4/toolkit.py", line 141, in ui_subpanel
    ui_panel.ui_subpanel( ui, parent )
root                                    : 2018-05-22 13:46:10,622 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traitsui/qt4/ui_panel.py", line 76, in ui_subpanel
    _ui_panel_for(ui, parent, True)
root                                    : 2018-05-22 13:46:10,623 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traitsui/qt4/ui_panel.py", line 82, in _ui_panel_for
    ui.control = control = _Panel(ui, parent, is_subpanel).control
root                                    : 2018-05-22 13:46:10,623 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/launchers/helpers.py", line 138, in __init__
    self.control = panel(ui)
root                                    : 2018-05-22 13:46:10,623 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traitsui/qt4/ui_panel.py", line 255, in panel
    panel = _GroupPanel(content[0], ui).control
root                                    : 2018-05-22 13:46:10,623 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traitsui/qt4/ui_panel.py", line 592, in __init__
    layout = self._add_items(content, inner)
root                                    : 2018-05-22 13:46:10,623 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traitsui/qt4/ui_panel.py", line 860, in _add_items
    editor.prepare(inner)
root                                    : 2018-05-22 13:46:10,623 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traitsui/editor.py", line 158, in prepare
    self.update_editor()
root                                    : 2018-05-22 13:46:10,623 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traitsui/qt4/instance_editor.py", line 319, in update_e [... cut ...]
root                                    : 2018-05-22 13:46:10,625 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traits/trait_handlers.py", line 110, in _undefined_get
    name, class_of( object ) ))
root                                    : 2018-05-22 13:46:10,625 CRITICAL  (MainThread) traits.trait_errors.TraitError: The 'preferred_ages' trait of an InterpretedAgeGroup instance is a property that has no 'get' or 'set' method
root                                    : 2018-05-22 13:46:10,625 CRITICAL  (MainThread) ============ End Exception ==========
ignore exception
```

wcmcintosh, 2018-05-22:
Tested.  Got further, interpret age window opened and looked fine, button got this error

```
Traceback (most recent call last):
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/ui/qt4/action/action_item.py", line 346, in _qt4_on_triggered
    self.controller.perform(action, action_event)
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/tasks/action/task_action_controller.py", line 31, in perform
    return action.perform(event)
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/tasks/action/listening_action.py", line 58, in perform
    method()
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 324, in set_interpreted_age
    set_interpreted_age(self.dvc, ias)
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/interpreted_age_factory.py", line 99, in set_interpreted_age
    dvc.add_interpreted_age(ia)
  File "/Users/mcintosh/Programming/git/pychron/pychron/dvc/dvc.py", line 1013, in add_interpreted_age
    ages=ia.ages(),
  File "/Users/mcintosh/Programming/git/pychron/pychron/processing/analyses/analysis_group.py", line 604, in ages
    vs = {k: getattr(self, k) for k in ('weighted_mean', 'plateau', 'isochron', 'integrated')}
  File "/Users/mcintosh/Programming/git/pychron/pychron/processing/analyses/analysis_group.py", line 604, in <dictcomp>
    vs = {k: getattr(self, k) for k in ('weighted_mean', 'plateau', 'isochron', 'integrated')}
AttributeError: 'InterpretedAgeGroup' object has no attribute 'weighted_mean'
```

wcmcintosh, 2018-05-22:
new error on setting interpreted age from ideogram

```
Traceback (most recent call last):
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/ui/qt4/action/action_item.py", line 346, in _qt4_on_triggered
    self.controller.perform(action, action_event)
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/tasks/action/task_action_controller.py", line 31, in perform
    return action.perform(event)
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/tasks/action/listening_action.py", line 58, in perform
    method()
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 324, in set_interpreted_age
    set_interpreted_age(self.dvc, ias)
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/interpreted_age_factory.py", line 99, in set_interpreted_age
    dvc.add_interpreted_age(ia)
  File "/Users/mcintosh/Programming/git/pychron/pychron/dvc/dvc.py", line 1023, in add_interpreted_age
    ia.analyses])
  File "/Users/mcintosh/Programming/git/pychron/pychron/dvc/dvc.py", line 1022, in <listcomp>
    tag=ai.tag, plateau_step=ia.get_is_plateau_step(ai)) for ai in
  File "/Users/mcintosh/Programming/git/pychron/pychron/processing/isotope_group.py", line 423, in __getattr__
    raise AttributeError(attr)
AttributeError: kcal
```

wcmcintosh, 2018-05-23:
"Saving interpreted age" operating without error, but saved values are NaN
[5.pdf](https://github.com/NMGRL/pychron/files/2032443/5.pdf)

wcmcintosh, 2018-05-23:
Set interpreted error now running without errors.
However, error saved is SEM even if SEM * root MSWD is selected.
 Interpreted age window shows SEM * root MSWD as selection.
The good news is that interpreted age ideogram seems to be working fine.

right click delete button in interpreted age selection pane not causing deletion

wcmcintosh, 2018-05-24:
Should interpreted ages be auto-populated to match source figure? 
Currently SEM only, non-responsive to age error type changes.

wcmcintosh, 2018-05-25:
Interpreted ages saving correctly, but restart required to view
correct  error values saved, but recalled interpreted age shows wrong method (weighted mean, not weighted mean * root MSWD)

jirhiker, 2018-05-25:
You are misinterpreting what you are seeing
1. Could not replicate restart required
2. you cant "recall" an interpreted age but in the browser the Interpreted Age table has a column called "Kind" This refers to the age calculation not the error calculation. weighted mean * root MSWD is not something we do. Its SEM * root MSWD

wcmcintosh, 2018-05-25:
small IA glitch: where a an analyses group begins with a subgroup, the identifier field comes up blank when saving the IA.  Possibly a consequence of  subgroup names -01pl vs xxx-01pl? I don't think name can be typed into he identifier field.

wcmcintosh, 2018-05-25:
Delete working now..  I also could not replicate restart required to see new IAs. Thanks for adding error kind to IA browser.  Thanks for widening columns

closed 2018-05-25 by jirhiker