NMGRL/pychron #1492: Table issues

wcmcintosh, 2018-05-22:
Tables working well with grouping and subgrouping, and requested summary ages are being calculated

A few display issues remain:  
Human table 
- could mswd be added to weighted mean and plateau display lines, as MSWD=x.xxx after n=x/y?
-could sensitivity be formatted in scientific notation?
-could vertical cell border between columns b and C be removed?
-could weighted mean have 1 vs 2 sigma option (option fine in summary table?
-could %39 be included for omitted steps?

Summary table
-for mean ages,  n column currently displays nt.  Could it display n/nt, like in the human table
-for a=1,s>1 subgroups, reported n=1 and reported MSWD=0.  Human table has correct values

jirhiker, 2018-05-22:
give mean an example excel file with the changes you want highlighted

wcmcintosh, 2018-05-22:
here is highlighted example, except for 1/2 sigma option for human table configuration
[1.xlsx](https://github.com/NMGRL/pychron/files/2028464/1.xlsx)

wcmcintosh, 2018-05-23:
summary table changes tested - OK
1 vs 2 sigma tested  - OK
other items have small problems, see attached excel file
[3.xlsx](https://github.com/NMGRL/pychron/files/2032075/3.xlsx)

wcmcintosh, 2018-05-23:
tested - all issues in 3.xls fixed except one miserable cell border.  see "border gap" in 4.xls
[4.xlsx](https://github.com/NMGRL/pychron/files/2032375/4.xlsx)

wcmcintosh, 2018-05-23:
I should have noticed sooner, but weighted-mean age errors in tables are not multiplied by sqrt-mswd, leading to  table/figure error mismatch.  
Other table features inc subgrouping etc. working perfectly.

jirhiker, 2018-05-23:
controlling the subgrouping error kind was never requested

wcmcintosh, 2018-05-23:
Like I say, I should have noticed and requested sooner.  My mistake.

wcmcintosh, 2018-05-24:
Attached excel file has 4 high-lighted small problems.  Thanks.
[9.xlsx](https://github.com/NMGRL/pychron/files/2036433/9.xlsx)

wcmcintosh, 2018-05-24:
All four problems resolved.  Ready to close.

closed 2018-05-24 by jirhiker