NMGRL/pychron #1494: Improve table footnotes - job for Bill and both Matts

wcmcintosh, 2018-05-22:
Bill and both Matts should work on a consensus version of notes for human and summary tables

jirhiker, 2018-05-23:
added Pychron/setupfiles/pipeline/table_notes.yaml so that table notes can be easily copied between computers.

table_notes.yaml is simple dict format
```yaml
blank_notes:  """asfasfafafaf asfasfaf asfasfasfa asfasfasfasf fasfsf"""
air_notes:  """asfasfafafaf asfasfaf asfasfasfa asfasfasfasf fasfsf"""
unknown_notes: """"dsafsfds"""
monitor_notes: """asdfasfasfasffafafafafafafaf"""
```

wcmcintosh, 2018-05-23:
Thanks Jake. 
I'll check this out.

wcmcintosh, 2018-05-23:
What values can the table notes access?  I see that unknown notes can access variables like decay_ref:

Decay constants and isotopic abundances after {decay_ref:}

wcmcintosh, 2018-05-23:
True that could occur, albeit rare in our lab.
Another analogous problem in the ancient macro version was caused by including irradiation interference  parameters in footnotes, requiring special treatment for multiple irradiations. Pychron includes this as columns within the data table and not in the footnotes.  
Does it make sense for footnotes to access decay_ref, iso_abundance_ref, monitor_age, etc from invisible columns in the data table?  Or is this a poor way to handle this?

jirhiker, 2018-05-23:
It seems like an exceedingly common mode of operation at other labs. 
decay_ref, monitor_age will be added as additional columns but that does not resolve which values are displayed in the footnotes.

wcmcintosh, 2018-05-23:
Attached is a draft of footnotes for standard pychron data tables.
I made these as one-size-fits all as I could. 
Hopefully OK as a start.
I don't know how live or dead you prefer to make these at the moment, in terms of decay, monitor age etc. 
I don't have a simple and effective solution for cases of multiple monitors etc.
In the short term, we could apply non-live footnotes and let the user modify one way or another.  being able to save and select note versions might be an idea in the future.
I'll work on footnotes for the summary table next.

[Footnotes 2.xlsx](https://github.com/NMGRL/pychron/files/2032664/Footnotes.2.xlsx)

wcmcintosh, 2018-05-23:
I didn't type this into pychron yet, just not excel to get your comments on the content first.

jirhiker, 2018-05-23:
looks fine except for 
1. Weighted-mean or plateau age error is inverse-variance-weighted mean error (Taylor, 1982) times root MSWD where MSWD>1. Not strictly true
2. missing description for "pX"

wcmcintosh, 2018-05-23:
"Not strictly true".  Not true because the user can select otherwise? There are a quite a number of items in the footnotes that could be auto-customized according to user selection, but I thought these could be addressed at a later date.
e.g. custom-matched mean calculation method, decay, abundances, monitor ages
     plateau footnotes only included if plateau data in table, etc.
But I figured no rush on any of this 

How does this look for a pX footnote?
[Footnotes 3.xlsx](https://github.com/NMGRL/pychron/files/2032826/Footnotes.3.xlsx)

wcmcintosh, 2018-05-26:
Modified footnotes by changing  file xlsx_table_options.py.   Hopefully this is the correct place.  Will test when table making running see #1499 error

jirhiker, 2018-05-26:
its not necessary to modify the source to modify your table footnotes. 

see comment from a few days ago in this thread. re `../setupfiles/pipeline/table_notes.yaml`

wcmcintosh, 2018-05-26:
When I look here ../setupfiles/pipeline/table_notes.yaml, I don't find the file.  Was this implemented? Or is it just not on my computer?  They only place I can locate the notes on my computer is in the code within xlsx_table_options.py

jirhiker, 2018-05-26:
you have to create the file

jirhiker, 2018-05-26:
```yaml
unknown_notes: |
  Errors quoted for individual analyses include analytical error only, without interfering reaction or J uncertainties.
  Weighted-mean or plateau age is inverse-variance-weighted mean of included analyses.
  Weighted-mean or plateau age error is inverse-variance-weighted mean error (Taylor, 1982) times root MSWD where MSWD>1.
  Integrated age calculated by summing isotopic measurements of all steps.
  Weight percent K2O calculated from 39Ar signal, sample weight, and instrument sensitivity.
  Decay constants after Min et al. (2000), LambdaK (total) =  5.464e-10/a
  Isotopic abundances after Steiger and Jäger (1977)
  Ages calculated relative to FC-2 Fish Canyon Tuff sanidine interlaboratory standard at 28.201 Ma after Kuiper et al. (2008)
  
blank_notes: 
monitor_notes:
air_notes:

```

wcmcintosh, 2018-05-26:
I've been rethinking footnotes a little.  See attached file.
This groups the footnotes differently than before, and includes much more info on instrumentation, etc.  I have been using similar footnotes in published summary tables for years now.

In the attached version, all lines would go into the human table, and all but yellow lines would go into the summary table.  There are a few conditional lines, dependent on whether plateau ages are present in the table and whether K20 values are present.  Conditionals could wither be handled dynamically, or perhaps footnotes could have standard-style multiple configurations, with save, save as, etc. 

Can give me guidance on formatting subscripts, superscripts, and italics (highlighted) or does this need to be hardcoded?

I can put these directly into Table notes, but I thought that I would run this by you first to get your thoughts.
[Footnotes 11.docx](https://github.com/NMGRL/pychron/files/2041960/Footnotes.11.docx)

jirhiker, 2018-05-26:
```yaml
unknown_notes: |
  this is a note with super/sub scripts <sup>39</sup>Ar <sup>K</sup>
```

wcmcintosh, 2018-05-28:
Tested.  <sup>  superscript working fine. 
<sub> subscript not working
could use <italics> and <bold>

I'm working on footnotes txt.  I currently have 3 versions:

1) unknown table, for data with no plateaus
2)  unknown table, for data with plateaus
3) summary table (similar to unknown but with no reference to individual analyses)

The two unknown footnote sets would start with the pychron generated footnote lines. Not needed for the summary table.

Not sufre about how to handle some the footnote choices.  Automated choice? Multiple configurations to choose from?

[footnote_test.xlsx](https://github.com/NMGRL/pychron/files/2046234/footnote_test.xlsx)

jirhiker, 2018-05-29:
```yaml
blank_notes: 
monitor_notes:
air_notes:
unknown_notes:
  Default: 
    main: |
        Notes and abbreviations: n/nt = number of analyses use to compute age/total analyses, MSWD = Mean Square of Weighted Deviates.
        Sample preparation: Minerals were separated by crushing and sieving, followed by etching in 7% HF (sanidine only) or 7% HCl and rinsing in an ultrasonic bath, then concentrated using a Franz magnetic separator and/or density liquids, followed by hand picking.
        Irradiation: Samples and Fish Canyon Tuff sanidine monitors in machined aluminum trays were irradiated <ital>in-vacuo</ital> at the Denver USGS Triga reactor.
        Instrumentation and procedures: Samples heated using a defocused CO <sub>2</sub> laser for 30 seconds. Reactive gases removed during analysis by reaction with a SAES NP-10 getter. Analyses were performed using Thermo Argus VI and/or Helix mass spectrometers on line with automated all-metal extraction system.
        J-factors were determined by CO <sub>2</sub> laser-fusion of 4 to 6 single crystals from multiple radial positions around the irradiation tray.
        Age and error calculations: Mean age errors quoted as <plus_minus>2<sigma>, all other errors <plus_minus>1<sigma>. Calculations performed using Pychron software, version xxx. Weighted-mean and error are inverse-variance-weighted mean and error Taylor (1982), with error multiplied by root MSWD where MSWD>1.
        Constants: Decay constants after Min et al. (2000), <lambda> K <sub>total</sub> = 5.464e-10/a. Isotopic abundances after Steiger and Jäger (1977). Ages calculated relative to FC-2 Fish Canyon Tuff sanidine interlaboratory standard at 28.201 Ma after Kuiper et al.(2008).
    
    corrected: |
      Corrected: Isotopic intensities corrected for blank, baseline,radioactivity decay and detector intercalibration, not for interfering reactions.
    
    x: X symbol preceding sample ID denotes analyses excluded from weighted-mean age calculations.
    
    px: pX symbol preceding sample ID denotes analyses excluded plateau age calculations.
    
    intercept: |
      Intercepts: t-zero intercept corrected for detector baseline.
    
    time: |
      Time interval (days) between end of irradiation and beginning of analysis.
      
  With Plateaus: 
    main: |
      Notes and abbreviations: n/nt = number of analyses use to compute age/total analyses, MSWD = Mean Square of Weighted Deviates.
      Sample preparation: Minerals were separated by chrushing and sieving, followed by etching in 7% HF (sanidine only) or 7% HCl and rinsing in an ultrasonic bath, then concentrated using a Franz magnetic separator and/or density liquids, followed by hand picking.
      Irradiation: Samples and Fish Canyon Tuff sanidine monitors in machined aluminum trays were irradiated <ital>in-vacuo</ital> at the Denver USGS Triga reactor.
      Instrumentation and procedures: Samples heated using a defocused CO<sub>2</sub> laser for 30 seconds. Reactive gases removed during analysis by reaction with a SAES NP-10 getter. Analyses were performed using Thermo Argus VI and/or Helix mass spectrometers on line with automated all-metal extraction system. J-factors were determined by CO <sub>2</sub> laser-fusion of 4 to 6 single crystals from each of multiple radial positions around the irradiation tray.
      Age and error calculations: Mean age errors quoted as <plus_minus>2<sigma>, all other errors <plus_minus>1<sigma>. Calculations performed using Pychron software, version xxx. Weighted-mean or plateau age and error are inverse-variance-weighted mean and error Taylor (1982), with error multiplied by root MSWD where MSWD>1. Plateau steps selected using criteria of Fleck et al. (1977). Cum.% <sup>39</sup> Ar is cumulative percent of total <sup>39</sup>Ar used in weighted-mean or plateau age calculations.
      Constants: Decay constants after Min et al. (2000), <lambda> K <sub>total</sub> = 5.464e-10/a. Isotopic abundances after Steiger and Jäger (1977).Ages calculated relative to FC-2 Fish Canyon Tuff sanidine interlaboratory standard at 28.201 Ma after Kuiper et al.(2008).
    
    corrected: |
      Corrected: Isotopic intensities corrected for blank, baseline,radioactivity decay and detector intercalibration, not for interfering reactions.
    
    x: X symbol preceding sample ID denotes analyses excluded from weighted-mean age calculations.
    
    px: pX symbol preceding sample ID denotes analyses excluded plateau age calculations.
    
    intercept: |
      Intercepts: t-zero intercept corrected for detector baseline.
    
    time: |
      Time interval (days) between end of irradiation and beginning of analysis.
```

closed 2018-06-05 by jirhiker