NMGRL/pychron #1495: Spectrum pipeline - plateau else mean

wcmcintosh, 2018-05-22:
Could Spectrum figure have "plateau else mean" capability similar to table subgrouping, such that if no plateau was found, the mean age (initially of all steps) would be displayed plateau style, but labelled appropriately.  I believe Jake has already taken steps to allow displaying mean ages in this fashion, but I don't think "plateau else mean" is available in the spectrum p[ipeline

Labelling-wise, one could argue that plateau ages are in fact weighted mean ages, so that labelling could be 
"Plateau-weighted-mean age =" vs. "No plateau- weighted-mean age =" or some such variation
or perhaps "Plateau" vs "weighted-mean" or "mean" is better and more succinct

jirhiker, 2018-05-23:
Hmm. the weighted mean age of all the valid steps can be displayed along with the integrated age.

wcmcintosh, 2018-05-23:
Weighted mean does display at the bottom of the screen, but displaying it like a plateau age (over a bar showing steps used) lends graphic clarity. I believe you had this feature working recently.  Am I not accessing it properly?
  
My thinking here is to use similar treatment for our various weighted-mean ages for selected steps, however selected ( machine applying arbitrary human criteria, human choice, etc).

jirhiker, 2018-05-23:
its more problematic in the mechanics/implementation than the concept

wcmcintosh, 2018-05-23:
Display of weighted-mean age as bar is working perfectly.
It appears that plateau age is displayed if present, otherwise, (if requested) weighted-mean is displayed.  This is exactly what I was looking for, without even having to specify "plateau else mean". Excellent.
Could plateau ages be prefixed with "Plateau="?  Thanks

wcmcintosh, 2018-05-23:
Two other questions on spectra. 
1) Ages ±xsigma and Err env=±y sigma both appear in the upper left corner of the lower panel.  "Err env" is set from the plateau options, and matches the display. "Ages" is set from the display option, and I not sure what it controls. Is it real or fossil?
2) What does the "general" checkbox in the display options control?

jirhiker, 2018-05-23:
1. Its not "Ages" its "Age" and it refers to the nsigma for display for the plateau/integrated/weighted mean age

2. uncheck it and find out. It simply turns off all messages displayed in upper left corner.

wcmcintosh, 2018-05-24:
Thanks Jake.  I see how these  work now.  Should it be "Ages", given that it refers to more than one (integrated, plateau, mean, etc.)  
Thanks also for adding the Plateau= label.
Getting back to the initial issue, I think the Spectrum pipeline is now handling plateau and other mean ages well.
Ready to close.