NMGRL/pychron #1498: error on diff view

nmgrluser, 2018-05-24:
active branch=release/py3/v18.2

active analyses=66123-01

description=""

Traceback
```
Traceback (most recent call last):
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.4/site-packages/pyface/ui/qt4/action/action_item.py", line 346, in _qt4_on_triggered
    self.controller.perform(action, action_event)
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.4/site-packages/pyface/tasks/action/task_action_controller.py", line 31, in perform
    return action.perform(event)
  File "/Users/argonlab2/miniconda2/envs/pychron3/lib/python3.4/site-packages/pyface/tasks/action/listening_action.py", line 58, in perform
    method()
  File "/Users/argonlab2/.pychron.1/updates/pychron/pipeline/tasks/task.py", line 185, in diff_analysis
    if editor.setup(left):
  File "/Users/argonlab2/.pychron.1/updates/pychron/pipeline/editors/diff_editor.py", line 237, in setup
    right = self._find_right(left)
  File "/Users/argonlab2/.pychron.1/updates/pychron/pipeline/editors/diff_editor.py", line 268, in _find_right
    left.step)
  File "/Users/argonlab2/.pychron.1/updates/pychron/mass_spec/mass_spec_recaller.py", line 49, in find_analysis
    rec.sync(dbrec)
  File "/Users/argonlab2/.pychron.1/updates/pychron/processing/analyses/analysis.py", line 441, in sync
    self._sync(obj, **kw)
  File "/Users/argonlab2/.pychron.1/updates/pychron/mass_spec/mass_spec_analysis.py", line 101, in _sync
    self.lambda_k = prefs.Lambda40Kepsilon + prefs.Lambda40KBeta
AttributeError: can't set attribute

```

wcmcintosh, 2018-05-25:
New diff view error

```
Traceback (most recent call last):
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/ui/qt4/action/action_item.py", line 346, in _qt4_on_triggered
    self.controller.perform(action, action_event)
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/tasks/action/task_action_controller.py", line 31, in perform
    return action.perform(event)
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/tasks/action/listening_action.py", line 58, in perform
    method()
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 186, in diff_analysis
    editor.set_diff(left)
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/editors/diff_editor.py", line 250, in set_diff
    self._set_values(left, right, isotopes)
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/editors/diff_editor.py", line 321, in _set_values
    ('Lambda Cl36', 'lambda_Cl36'))]
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/editors/diff_editor.py", line 318, in <listcomp>
    for n, k in (('Lambda K', 'lambda_k'),
  File "/Users/mcintosh/Programming/git/pychron/pychron/processing/isotope_group.py", line 423, in __getattr__
    raise AttributeError(attr)
AttributeError: lambda_Ar37
```

wcmcintosh, 2018-05-25:
Diff view working, ready to close.

closed 2018-05-25 by jirhiker