NMGRL/pychron #1499: generalized means for K/Ca, K/Cl, etc.

wcmcintosh, 2018-05-25:
Your idea of having a generalized approach to mean value for K/Ca, K/Cl, etc. (each with associated mean and error methods) seems very good, essentially putting K/Ca/ K/Cl etc (and age??) on equal  footing.  

One potential spinoff is being able to calculate and plot means for values like %rad, moles39, etc, facilitating plotting subgroup mean values on stacked ideograms, or using group mean values for comparisons among samples.

For %rad  and moles 39, integrated means might be the method to select, but we never had this capability before.

wcmcintosh, 2018-05-25:
Re:#1501 - Purely philosophically, does it make sense to term non-existent data as 0±0 and plot it thus with valid data?

jirhiker, 2018-05-25:
No it doesn't but my point is that non-existent data should not exist in the first place. I does not make sense to make the ideogram exclude analyses that don't have data when the reason the data doesn't exist is because it hasn't been implemented. 

Omitting analyses because a value doesn't exist (because it hasn't been implemented) just masks the real problem

wcmcintosh, 2018-05-25:
Again, from a purely philosophical standpoint, It is true that some datasets are incomplete for various reasons other than a feature not yet being, e.g. something not measured in some analyses.  Comparing various data sets of differing completeness can be worthwhile, omitting points as required where no valid data exists.  I bring this up more for future reference than for this specific case, which can be better addressed by implmentation.

jirhiker, 2018-05-25:
"omitting points as required where no valid data exists" This is the most contentious aspect. I think its bad practice to do this automatically because its very difficult to do so that the user is not confused

wcmcintosh, 2018-05-26:
Error in last node of table making ID#655761
error seemingly related to new mean changes?
```
ignore exception
Pipeline Data Processing                : 2018-05-26 08:31:03,942 DEBUG     (MainThread) resume pipeline started
pychronmeta                             : 2018-05-26 08:31:03,942 DEBUG     (MainThread) force create new session 4838745896
root                                    : 2018-05-26 08:31:03,958 CRITICAL  (MainThread) ============ Exception ==============
root                                    : 2018-05-26 08:31:03,958 CRITICAL  (MainThread) Traceback (most recent call last):
root                                    : 2018-05-26 08:31:03,959 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/ui/qt4/action/action_item.py", line 346, in _qt4_on_triggered
    self.controller.perform(action, action_event)
root                                    : 2018-05-26 08:31:03,959 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/tasks/action/task_action_controller.py", line 31, in perform
    return action.perform(event)
root                                    : 2018-05-26 08:31:03,959 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/tasks/action/listening_action.py", line 58, in perform
    method()
root                                    : 2018-05-26 08:31:03,959 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 314, in resume
    self._resume_pipeline()
root                                    : 2018-05-26 08:31:03,959 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 489, in _resume_pipeline
    self._run('resume pipeline', 'resume_pipeline')
root                                    : 2018-05-26 08:31:03,959 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 471, in _run
    if not getattr(self.engine, func)():
root                                    : 2018-05-26 08:31:03,959 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/engine.py", line 655, in resume_pipeline
    return self.run_pipeline(state=self.state)
root                                    : 2018-05-26 08:31:03,959 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/engine.py", line 722, in run_pipeline
    pipeline.resume(state)
root                                    : 2018-05-26 08:31:03,960 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/engine.py", line 97, in resume
    prev_node.resume(state)
root                                    : 2018-05-26 08:31:03,960 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/nodes/group_age.py", line 45, in resume
    ias = make_interpreted_age_subgroups(ans)
root                                    : 2018-05-26 08:31:03,960 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/subgrouping.py", line 105, in make_interpreted_age_subgroups
    preferred_age_error_kind=error_kind)
root                                    : 2018-05-26 08:31:03,960 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/processing/analyses/analysis_group.py", line 113, in __init__
    super(AnalysisGroup, self).__init__(make_arar_constants=False, *args, **kw)
root                                    : 2018-05-26 08:31:03,960 CRITICAL  (MainThread) File "/Users/mcintosh/Programming/git/pychron/pychron/processing/analyses/analysis.py", line 185, in __init__
    super(IdeogramPlotable, self).__init__(*args, **kw)
root                                    : 2018-05-26 08:31:03,960 CRITICAL  (MainThread) File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traits/trait_handlers.py", line 173, in error
    value )
root                                    : 2018-05-26 08:31:03,960 CRITICAL  (MainThread) traits.trait_errors.TraitError: The 'preferred_age_kind' trait of an InterpretedAgeGroup instance must be 'Weighted Mean' or 'Integrated' or 'Arithmetic Mean' or 'Plateau' or 'Isochron', but a value of 'Plateau else Weighted Mean' <class 'str'> was specified.
root                                    : 2018-05-26 08:31:03,960 CRITICAL  (MainThread) ============ End Exception ==========
ignore exception
```

wcmcintosh, 2018-05-26:
Subgroup ideogram yields same error in final node

wcmcintosh, 2018-05-26:
#1509 bugs probably related to generalized means.

wcmcintosh, 2018-05-26:
Generalized means running well in table and ideogram.  Excellent.Thanks a lot.
%rad and 39Ar values correct ballpark.
Only tested so far with all weighted means.  Are the other means available now? I.e. integrated %rad and 
Could new values be displayed on subgroup and group mean lines, either as xxx= or in appropriate column?
Sub group point tables not displaying we, pl, etc, although -01we show up in hover window.

jirhiker, 2018-05-26:
other means are available. 

"Sub group point tables not displaying we, pl" Again you have to label the points with "label_name" not "aliquot"

wcmcintosh, 2018-05-26:
OaK. Got it.  Aliquots 01 and 02 correctly displaying as 01we, 02pl, etc.  Only very minor problem is that 03-09 are displaying as 3-9 in this mode, as opposed to 03-09 in aliquot mode.

wcmcintosh, 2018-05-27:
I'll test the other means next, but I will have to do it later this evening,

wcmcintosh, 2018-05-28:
Interactive means  within set interpreted age window very nice.  Should error method also be included?
Mean values all appear to be stored correctly with IA.  However, Ideogram IA still plotting %rad and 39Ar as zero.  Will check subgroup pipelines.

wcmcintosh, 2018-05-28:
Plotting generalized mean valiues:
Have you made provisions to include generalized mean values in mean lines in tables? It seems like generalized means are being calculated correctly and stored with IAs, but not yet included in the tables. Currently can't run subgroup ideogram due to error, but generalized mean values were previously plotting fine form subgroup ideogram.  Plotting as zero from IA ideogram.

wcmcintosh, 2018-05-29:
Subgroup ideogram is plotting means fine.

wcmcintosh, 2018-06-01:
Various generalized means working well for subgroups.  Not fully implemented for groups, i.e. mean %rad and 39Ar not yet displayed, but that can come later.  Ready to close.

closed 2018-06-01 by jirhiker