NMGRL/pychron #1500: No Title Provided

wcmcintosh, 2018-05-25:
active branch=None

active analyses=66196-01,66196-02,66196-03,66196-04,66196-05,66196-06,66196-07,66196-08,66196-09

description=""

Traceback
```
Traceback (most recent call last):
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/ui/qt4/action/action_item.py", line 163, in _qt4_on_triggered
    self.controller.perform(action, action_event)
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traitsui/qt4/editor.py", line 269, in perform
    self.ui.do_undoable( self._perform, action )
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traitsui/ui.py", line 716, in do_undoable
    action( *args, **kw )
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traitsui/qt4/editor.py", line 291, in _perform
    method( info, selection )
  File "/Users/mcintosh/Programming/git/pychron/pychron/envisage/browser/sample_view.py", line 392, in delete
    obj.delete()
  File "/Users/mcintosh/Programming/git/pychron/pychron/envisage/browser/interpreted_age_table.py", line 63, in delete
    os.remove(r.path)
FileNotFoundError: [Errno 2] No such file or directory: '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Hanneman01074/661/ia/96.ia.json'

```

jirhiker, 2018-05-25:
not clear how this could happen. fix may be masking some other problem

wcmcintosh, 2018-05-25:
Jake:Could you clarify what you mean by "not clear how this could happen".  I did type an informative title for this but didn't click out of the box, and now I don't remember just what triggered this.

I quite like the concept of "do_undoable". I  bet it could be worked into a song title.

jirhiker, 2018-05-25:
`FileNotFoundError: [Errno 2] No such file or directory: '/Users/mcintosh/Pychron_dev/data/.dvc/repositories/Hanneman01074/661/ia/96.ia.json'`
happens when pychron is trying to delete the ia file. The error says the file doesn't exist. BUT the only way pychron knows to delete the file is because it found it. So the search finds the file but when trying to delete the found file python cant find it.

closed 2018-05-26 by jirhiker