NMGRL/pychron #1507: error on startup after update

wcmcintosh, 2018-05-26:
```
Traceback (most recent call last):
  File "/Users/mcintosh/Programming/git/pychron/pychron/envisage/pychron_run.py", line 246, in launch
    app.run()
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/envisage/ui/tasks/tasks_application.py", line 140, in run
    self._create_windows()
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/envisage/ui/tasks/tasks_application.py", line 305, in _create_windows
    restore=self.always_use_default_layout)
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/envisage/ui/tasks/tasks_application.py", line 214, in create_window
    task = self.create_task(task_id)
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/envisage/ui/tasks/tasks_application.py", line 167, in create_task
    task = factory.create_with_extensions(extensions)
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/envisage/ui/tasks/task_factory.py", line 33, in create_with_extensions
    task = self.create(**traits)
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/envisage/ui/tasks/task_factory.py", line 28, in create
    return self.factory(**traits)
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/plugin.py", line 82, in _pipeline_factory
    from pychron.pipeline.tasks.task import PipelineTask
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 34, in <module>
    from pychron.pipeline.engine import PipelineEngine
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/engine.py", line 29, in <module>
    from pychron.pipeline.nodes import FindReferencesNode
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/nodes/__init__.py", line 42, in <module>
    from pychron.pipeline.nodes.persist import DVCPersistNode, PDFFigureNode, \
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/nodes/persist.py", line 42, in <module>
    from pychron.pipeline.tables.xlsx_table_options import XLSXAnalysisTableWriterOptions
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tables/xlsx_table_options.py", line 76
    unknown_notes = dumpable(Str('''Errors quoted for individual analyses include analytical error only, without interfering reaction or J uncertainties.
                                                                                                                                                        ^
TabError: inconsistent use of tabs and spaces in indentation
2018-05-26 09:58:43.781 python[75586:11340461] modalSession has been exited prematurely - check for a reentrant call to endModalSession:

```

wcmcintosh, 2018-05-26:
Still have error after next update,

jirhiker, 2018-05-26:
you broke `xlsx_table_options.py` when you edited it

wcmcintosh, 2018-05-26:
I just realized that.  I will restore the original.

wcmcintosh, 2018-05-26:
Back in business.

closed 2018-05-26 by wcmcintosh