NMGRL/pychron #1508: Bug in Excel node of table

wcmcintosh, 2018-05-26:
active branch=None

active analyses=65761-01A,65761-01B,65761-01C,65761-01D,65761-01E,65761-02A,65761-02B,65761-02C,65761-02D,65761-02E,65761-03,65761-04,65761-05,65761-06,65761-07,65761-08,65761-09,65761-10,65761-11,65761-12,65761-13,65761-14,65761-15

description=""

Traceback
```
Traceback (most recent call last):
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/ui/qt4/action/action_item.py", line 346, in _qt4_on_triggered
    self.controller.perform(action, action_event)
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/tasks/action/task_action_controller.py", line 31, in perform
    return action.perform(event)
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/pyface/tasks/action/listening_action.py", line 58, in perform
    method()
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 314, in resume
    self._resume_pipeline()
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 489, in _resume_pipeline
    self._run('resume pipeline', 'resume_pipeline')
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 471, in _run
    if not getattr(self.engine, func)():
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/engine.py", line 655, in resume_pipeline
    return self.run_pipeline(state=self.state)
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/engine.py", line 747, in run_pipeline
    node.run(state)
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/nodes/persist.py", line 256, in run
    writer.build(state.groups, options=self.options)
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tables/xlsx_table_writer.py", line 77, in build
    self._make_human_unknowns(unknowns)
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tables/xlsx_table_writer.py", line 387, in _make_human_unknowns
    self._make_sheet(unks, 'Unknowns')
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tables/xlsx_table_writer.py", line 500, in _make_sheet
    age, label = ag.get_age(kind, set_preferred=True)
  File "/Users/mcintosh/Programming/git/pychron/pychron/processing/analyses/analysis_group.py", line 711, in get_age
    self.preferred_age_kind = label
UnboundLocalError: local variable 'label' referenced before assignment

```

jirhiker, 2018-05-26:
you must have tried integrated age

wcmcintosh, 2018-05-26:
Fresh bug referenced in #1509

wcmcintosh, 2018-05-26:
Bug fixed. Ready to close.

closed 2018-05-26 by jirhiker