NMGRL/pychron #1510: Low priory Table formatting requests

wcmcintosh, 2018-05-26:
when time permits - no rush, could you look at simple format enhancements listed in attached file

in short:
Human table: 
   Increased sig.fig granularity in table configuration
   Two missing cell borders beneath J columns

Summary Table:
   adjust some column widths
   add sig figs for mean age/err and KCa/err

[Table Format cleanup.pdf](https://github.com/NMGRL/pychron/files/2042010/Table.Format.cleanup.pdf)

jirhiker, 2018-05-27:
It would have been much better to known all these requirements from the beginning. 

Adding all this functionality will surely require me to redesign how tables are constructed

wcmcintosh, 2018-05-27:
Sorry Jake, I held back on this formatting request until more functional issues were resolved.  I was hoping that this would require  small changes to the configuration routines, more or less adding a few more significant figure text boxes to the existing configuration routines, rather than redesign.

jirhiker, 2018-05-27:
more a general comment the specific to this situation. 

unfortunately the formatting code was getting messier. More granular sig figs was the last straw.

wcmcintosh, 2018-05-27:
Jake Thanks for adding this.  It is running without error requested sig figs not matching columns correctly.  See attached examples.
[1.pdf](https://github.com/NMGRL/pychron/files/2042822/1.pdf)
[1.xlsx](https://github.com/NMGRL/pychron/files/2042823/1.xlsx)

wcmcintosh, 2018-05-29:
Formatting working fine except one column escaped formatting.  40Ar*/39ArK doesn't respond to any formatting settings.  It could go with signal or default or get its own box, whichever you like.

wcmcintosh, 2018-06-01:
Formatting excellent now.  Ready to close.

closed 2018-06-01 by jirhiker