NMGRL/pychron #1513: No Title Provided

wcmcintosh, 2018-05-28:
active branch=None

active analyses=65761 03

description=""

Traceback
```
Traceback (most recent call last):
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traits/trait_notifiers.py", line 519, in _dispatch_change_event
    self.dispatch( handler, *args )
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traits/trait_notifiers.py", line 482, in dispatch
    handler( *args )
  File "/Users/mcintosh/Programming/git/pychron/pychron/pipeline/tasks/task.py", line 580, in _handle_recall
    self.recall(new)
  File "/Users/mcintosh/Programming/git/pychron/pychron/envisage/browser/browser_task.py", line 132, in recall
    ans = self.dvc.make_analyses(records)
  File "/Users/mcintosh/Programming/git/pychron/pychron/dvc/dvc.py", line 770, in make_analyses
    level = r.irradiation_level
AttributeError: 'DVCInterpretedAge' object has no attribute 'irradiation_level'

```

wcmcintosh, 2018-05-28:
Question:Is the browser the only way to get to a list of IAs? How about viewing IAs for a run ID?  Right click? double click?

wcmcintosh, 2018-06-13:
Currently no response to double click if trying to get more info on IA from browser pane.  Not sure if you still plan to implement this.  IA's continue to save and update fine.

jirhiker, 2018-06-13:
"Currently no response to double click if trying to get more info on IA from browser pane"
This not implemented nor ever requested. 

"How about viewing IAs for a run ID" this is a very expensive operation as every interpreted age needs to be examined to see if it contains the selected RunID.

closed 2019-03-17 by jirhiker