NMGRL/pychron #1519: Error when dropdown in IA window changed.

wcmcintosh, 2018-05-29:
active branch=None

active analyses=65761 06

description=""

Traceback
```
Traceback (most recent call last):
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traits/trait_notifiers.py", line 519, in _dispatch_change_event
    self.dispatch( handler, *args )
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traits/trait_notifiers.py", line 482, in dispatch
    handler( *args )
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traits/has_traits.py", line 966, in wrapper0
    return function(arg)
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traits/has_traits.py", line 3372, in notify
    self.trait_property_changed( name, old )
  File "/Users/mcintosh/miniconda3/envs/pychron3/lib/python3.5/site-packages/traits/has_traits.py", line 896, in decorator
    self.__dict__[ name ] = result = function( self )
  File "/Users/mcintosh/Programming/git/pychron/pychron/processing/analyses/analysis_group.py", line 596, in _get_plateau_age
    k39 = [nominal_value(ai.get_computed_value('k39')) for ai in ans]
  File "/Users/mcintosh/Programming/git/pychron/pychron/processing/analyses/analysis_group.py", line 596, in <listcomp>
    k39 = [nominal_value(ai.get_computed_value('k39')) for ai in ans]
AttributeError: 'DVCInterpretedAge' object has no attribute 'get_computed_value'

```

jirhiker, 2018-05-29:
you are trying to set an interpreted age of interpreted ages?

wcmcintosh, 2018-05-29:
IA window working well now.  Only problem is mol 39Ar 

for 65761. - weighted mean mol 39Ar  is nan/nan, arithmetic mean is 0/0

wcmcintosh, 2018-05-29:
Closing this issue, opening a new one.

closed 2018-05-29 by wcmcintosh

wcmcintosh, 2018-05-29:
see #1520