NMGRL/pychron #1527: Standardize 39Ar units

wcmcintosh, 2018-05-30:
Ideally fA or moles could be chosen for 39A units.  
Tables are currently fA, but ideograms are moles.

Could we temporarily standardized of fA, so tables and figures match?  

Thanks

wcmcintosh, 2018-05-30:
That's great, but I haven't yet figured out how to switch

wcmcintosh, 2018-05-30:
I hit the bomb.  Other than the Pycharm message below, I don't see an effect.  Should the dropdown offer two 39Ar options?

wcmcintosh, 2018-05-30:
IdeogramOptionsManager                  : 2018-05-30 09:40:33,980 DEBUG     (MainThread) set factory default

jirhiker, 2018-05-30:
when i hit the bomb the options are cleared and the aux plots needed to be repopulated. 

There should be Signal 39Ar and Mol 39Ar

wcmcintosh, 2018-05-30:
I rebombed after last update, but no luck.  Heading into town now.

jirhiker, 2018-05-30:
setting the factory defaults only works for the default options "screen" and "presentation"

If not using either of those you will need to save a new options set

wcmcintosh, 2018-05-30:
mol39/fA39 working perfectly in figures. Ready to close.

closed 2018-05-30 by jirhiker