NMGRL/pychron #1528: request - change defaults in subgroup mean window

wcmcintosh, 2018-05-30:
Given the 99% argument, could the default mean/err types in the subgroup window be set for single-aliquot data?

[6.pdf](https://github.com/NMGRL/pychron/files/2054799/6.pdf)

jirhiker, 2018-05-30:
again this is trickier than you realize as the mean types are dependent on the number "groupings" but the number of groupings is not yet defined. Its being defined at this point. So this feature really doesn't fit into the current pipeline implementation well and will required not a trivial amount of modification. 

This precisely why it had not been implemented to date

wcmcintosh, 2018-05-30:
OK, I'll keep clicking away.

jirhiker, 2018-05-30:
in reality pychron was already automatically setting the defaults correctly. 

The subsequent node "Group Age" sets the default types

wcmcintosh, 2018-06-01:
Thanks for setting the default mean and error types.  Default age mean type in subgroup node is now set to weighted mean.  Any chance you could change it to "plateau else mean".  I know it is not your favorite, but now that it is working well, having it as default will handle all A==1 data gracefully.  Other than that, ready to close.

closed 2018-06-05 by jirhiker