NMGRL/pychron #1529: Footnotes request - no rush

wcmcintosh, 2018-05-30:
Request 1:
When convenient, could you add a section in table_notes.yaml for summary footnotes?
It could be similar to unknown notes with default: and If plateau:

I note that unknowns footnotes are currently not recognizing the presence of plateaus in subgroup IAs.

Request2:  could <bold><\bold> be added to footnotes. (It may already be there).  

I think sup, sub, ital are all working fine.

wcmcintosh, 2018-05-31:
<bold> and <sub> not operating consistently.xls file and table_notes.yaml attached
I've tried to work out why bold works in some places and not others, but I am not seeing the system.

[18.xlsx](https://github.com/NMGRL/pychron/files/2060498/18.xlsx)
[Table_notes.txt](https://github.com/NMGRL/pychron/files/2060505/Table_notes.txt)

jirhiker, 2018-06-01:
the problem is the space.

`<bold>Sample preparation</bold>`breaks my naive tag parsing implementation.  Ill have to come up with a solution. `<bold>AbcDef</bold>` will work `<bold>Abc Def</bold>` will not

multiline rendering should be possible. Ill check to see why its not working

wcmcintosh, 2018-06-01:
As a work around, I could mark each word, rather than each phrase.

Why do you think that the summary notes seem to be ignoring line breaks present in the Table_notes.yaml file?

closed 2018-06-05 by jirhiker