NMGRL/pychron #1530: mean value mismatch

wcmcintosh, 2018-05-30:
Some subgroup ideograms are giving different mean values than the table, slightly.  
See 65671 in attached table and figures.  I will investigate why.  Regular ideograms agree perfectly,and until recently so did 65671 subgroup ideogram.  It is probably something obvious.

[Streck2018.pdf](https://github.com/NMGRL/pychron/files/2055001/Streck2018.pdf)
[Streck2018.xlsx](https://github.com/NMGRL/pychron/files/2055002/Streck2018.xlsx)

jirhiker, 2018-05-30:
Those files are not a comparison of the same data

wcmcintosh, 2018-05-31:
You are right.  I see what's happening.  Spectrum and Table are correctly applying plateau else weighted-mean, and correctly selecting weighted-mean
SubGroup Ideogram is not correctly applying plateau else weighted-mean, instead calling everything plateaus.  Let me look at the plateau criteria.  Not sure where they are set for subgroups.

jirhiker, 2018-05-31:
sorry i meant that the xls was for 65733 and the ideo was for 65761 but i didn't realize there were multiple ideos in that pdf

jirhiker, 2018-05-31:
"Let me look at the plateau criteria. Not sure where they are set for subgroups." 
ugh. 

One more reason why the original explicit interpreted age model was so much more robust. Make a spectrum, choose plateau criteria, save as interpreted age. Use save interpreted ages to make subsequentt tables and figures

jirhiker, 2018-05-31:
another problem trying to use construction for replication instead of reproduction

wcmcintosh, 2018-05-31:
I think all five samples are in both files.  65671 demonstrates the problem. It may be that there is no means to set plateau criteria in this pipeline.  Plateau Criteria would be an additional candidate for same global setting node approach. Certain items like mean type and plateau criteria are so rarely changed. 
  Honestly, Subgroup was formerly perfectly A-OK, apparently getting its plateau criteria from the ether.  
Construction is the first step toward reproduction and replication,I think you told me recently.

wcmcintosh, 2018-05-31:
I do get the analogy, but until we have reproduction, do we have any choice but to double construct (figure + table) when we want to reproduce?

Your favorite, plateau else mean, working fine in table, and working almost fine in subgroup ideogram.

Subgroup Ideogram  plateau recognition:  Correctly (matches table) identifies weighted mean in 65761, and plateau in 65761.  Incorrectly (doesn't match table) but selects weighted-mean for 65762.  Other than that all OK.

Table (see attached) works well for all except 65773.  Human table correctly selects weighted mean for mean age, but summary table selects integrated age for mean age. Not requested.

jirhiker, 2018-05-31:
fixed the 65762 issue. 

The 65733 issue is because you requested that if aliquots==1 than set default mean type to integrated. I guess this default setting should be skipped for "age"

wcmcintosh, 2018-06-01:
Thanks for making the change.  For aliquots==1, I was hoping to set the default age to "plateau else mean" and all other mean values to integrated.  In any case, summary table now matches machine table, so ready to close.

closed 2018-06-01 by jirhiker