NMGRL/pychron #1532: A few small discrepancies in the summary table.

wcmcintosh, 2018-06-01:
As detailed on the summary sheet in the attached spreadsheet, the unknown sheet and the figures are matching well out to the 3rd sig fig anyway.  However there are some small discrepancies in the summary sheet numbers.  Most look like rounding errors, but shouldn't the math and rounding be the same in the unknown and summary sheets? 
 For one sample 65762, the unknown sheet and figure are using identical values for plateau ages for -01 and -02, but the summary sheet is (I think) using Weighted-mean ages.  So are there some differences between procedures?  Lastly, the spectrum figure for 65773 labels the mean age as "plateau=" rather than "weighted mean=".   The numbers are correct, but gas is <50%, so not a plateau.

Really sorry to keep bugging you with such things - just trying to make the numbers match.  The problems all seem to be in Type 3 data. Type 1 and Type 2 data match out to at least the 6th decimal place.

I would think that matching the unknown sheet to the summary sheet would be easier than matching the figures, but this seems not to be the case.

[discrepency.xlsx](https://github.com/NMGRL/pychron/files/2060717/discrepency.xlsx)

jirhiker, 2018-06-01:
i don't think rounding errors are possible

jirhiker, 2018-06-01:
"For one sample 65762, the unknown sheet and figure are using identical values for plateau ages for -01 and -02" last commit should address this. 

"Lastly, the spectrum figure for 65773 labels the mean age as "plateau=" rather than "weighted mean=". The numbers are correct, but gas is <50%, so not a plateau."
don't know about this but should be simple

wcmcintosh, 2018-06-01:
Hallelujah, preferred ages for tables (unknowns and summary) and figures all agree to six significant figures.  How about that?

wcmcintosh, 2018-06-01:
"Lastly, the spectrum figure for 65773 labels the mean age as "plateau=" rather than "weighted mean=". The numbers are correct, but gas is <50%, so not a plateau."
don't know about this but should be simple

The integrated age is n=4, therefore not including omitted steps, which it should.  I wonder if the plateau criteria are not being applied to all the steps, just the non-omitted steps.

wcmcintosh, 2018-06-01:
Problem corrected, ready to close. Moving "plateau=" problem to new issue.